amymcgovern/pyparrot #107: Connecting to Anafi Drone

CodeNeedsCoffee, 2018-10-03:
So I know the documentation said Anafi support will come when the official SDK is released, but I wasn't sure if anyone has made any progress yet? I was able to make some form of connection to the drone but I don't think I have the right connection yet. Any form of help would be greatly appreciated!

Cheers!

amymcgovern, 2018-10-03:
Sorry, I really want an anafi but don't own it yet!  Somewhere on the development board for parrot, they posted that it should work with minor tweaks:

https://forum.developer.parrot.com/t/parrot-anafi-sdk/7508/20

find1dream, 2019-05-20:
@amymcgovern, thanks for your great library! Any progress for Anafi drone? I really want to use your library in my project. Any form of help would be greatly appreciated!

amymcgovern, 2019-05-20:
I should be able to support the anafi with their new SDK release a few weeks ago.  Only problem is that I don't own an anafi to test it.  It is on my agenda.  Been hoping they will have a summer sale so I can buy one!