amymcgovern/pyparrot #108: Problems in getting pictures

YonghyunRyu, 2018-10-04:
I have found there are two problems when I get photos from the drone.
The two function `get_latest_valid_picture` in droneVision.py and `get_groundcam_picture` in miniDrone.py.

When I run the example program many times, they show or give me the picture the drone took in the last execution.
The main reason is the way that the source program brings photos to the programmer.
The source program a picture from the drone  the source program downloads the images into files from the drone and read them from the file.
This is because of higher latency between the drone and computer than the source program executes, 
The program commands the drone to send what it is taking, and the program immediately try to read data, but it takes a little time to receive the image.  So, the program reads an old file and shows me those old pictures that the drone took in the last execution.

In order to solve this problem, I have modified the two source codes `miniDrone.py` and `droneVision.py` to delete all the images the drone took and saved before.

I had a hard time finding the reason and solution of this problem.
I hope you will fix this problem or allow and help me to fix this.

YonghyunRyu, 2018-10-04:
I have forked this project and modified some files.
I have added the two lines below after the line 100 in `DroneVision.py`
```
cmd = "rm " + self.imagePath + "/image_*.png" 
os.system(cmd) 
```

amymcgovern, 2018-10-04:
Out of curiosity, are you running windows?  It has not been an issue on Mac or linux but I don't have a windows machine to test on.

YonghyunRyu, 2018-10-04:
I execute it on Ubuntu GNOME 16.04.

amymcgovern, 2018-10-15:
Sorry I was slow to get back on this.  I don't have linux by default (I develop on Mac).  But I was using the vision on linux this weekend and it was running fine.