amymcgovern/pyparrot #109: Parrot Mambo not connecting via wifi

Encouver, 2018-10-11:
Hello, the parrot Mambo is not connecting and throws the following message:

(pyparrot) C:\Users\Marcos\Desktop\DemoPyparrot>python demoMamboTricks.py
OpenCVAvailable is False
FTP login success is 230 Operation successful
82
C:\Users\Marcos\AppData\Local\conda\conda\envs\pyparrot\lib\site-packages\pyparrot\images\groundcam.jpg
trying to connect
[38;5;196m connection failed: did you remember to connect your machine to the Drone's wifi network? [0m
connected: False

This is from the demoMamboTricks example.

amymcgovern, 2018-10-11:
did you try it a second time?  Sometimes the wifi is just flakey on the mambo.  If it is doing it over and over again, my next question is:  are you running python 3 (it just says python there and it looks like you are using windows).

Encouver, 2018-10-11:
I've tried several times, and with the retries increased. I am running python 3  as you can see, in windows the executable for python 3 is just python:

(pyparrot) C:\Users\Marcos>python -V
Python 3.7.0

amymcgovern, 2018-10-11:
This is when I wish I could see the computer in question!  I've certainly observed the behavior and usually a reboot fixes things.  Rebooting the drone and windows.  It occurs much more frequently in windows than in any other OS, which makes me wonder if windows is hiding some permissions issues somewhere.

Encouver, 2018-10-12:
The reboot fixed it.

Issue closed.

closed 2018-10-12 by Encouver