amymcgovern/pyparrot #110: Running DroneVisionGUI from another QApplication

Spareman, 2018-10-11:
Hello there. This library has been great and I have to thank all the developers involved on that.
I have this issue:
Is it possible to call the DroneVisionGUI from another PyQT application and have both windows running? I am working on an app that has its own main window and I want to connect to the drone (Parrot Mambo) and see its stream and then run the user_code by doing something like `mamboVision.run_user_code(mamboVision.vlc_gui.playbutton)`, but the code keeps crashing and I haven't figured out why yet. Could it be because of the open_video function being blocking?? I am using a thread to call it in parallel.

amymcgovern, 2018-10-11:
I was able to make a second window with the VLC guy running but it I did it using opencv's vision functions.  I haven't tried to make a second QT window but I don't see why that wouldn't work.  What I do know is problematic is trying to run a second GUI using threading, at least on Mac (which is what I use to develop).  Mac wants the GUI window to be the main thread and will throw lots of errors if it is not.

amymcgovern, 2018-10-11:
I used an opencv named window:

            # setup the extra window to draw the markers in
            cv2.namedWindow("MarkerStream")

amymcgovern, 2018-10-11:
so the main QT GUI would display and then this one would open up once the program ran (once you pressed run) and then I could display other things in it.  Now I was specifically trying to display vision stuff so that's why I used openCV

Spareman, 2018-10-13:
I've also used the OpenCV to display parts of the stream and it works fine. What I want to do though is to have a GUI that gives commands to the drone to be performed and at the same time to be able to see the drone's stream while flying.
The best way that I found to do it so far (though haven't tested yet), is by using a subprocess to call the droneVisionGUI from my other GUI and passing the commands through PIPE to be performed.