amymcgovern/pyparrot #111: ftplib.error_perm: 530 Login failed

ihsancemil, 2018-10-19:
When I try demo-of-the-trick-commands-on-the-mambo this demo on linux-mint I got 

```
OpenCVAvailable is True
Traceback (most recent call last):
  File "examplefly.py", line 12, in <module>
    mambo = Mambo(None, use_wifi=True)
  File "/home/ihsancemil/.local/lib/python3.5/site-packages/pyparrot/Minidrone.py", line 310, in __init__
    self.groundcam = MamboGroundcam()
  File "/home/ihsancemil/.local/lib/python3.5/site-packages/pyparrot/Minidrone.py", line 237, in __init__
    login = self.ftp.login()
  File "/usr/lib/python3.5/ftplib.py", line 419, in login
    resp = self.sendcmd('PASS ' + passwd)
  File "/usr/lib/python3.5/ftplib.py", line 272, in sendcmd
    return self.getresp()
  File "/usr/lib/python3.5/ftplib.py", line 245, in getresp
    raise error_perm(resp)
ftplib.error_perm: 530 Login failed
```
I can read camera stream by my self so I know that my wifi connected to the mambo

`import cv2
import time

vcap = cv2.VideoCapture("rtsp://192.168.99.1/media/stream2 ! rtph264depay ! h264parse ! omxh264dec ! appsink max-buffers=1 drop=true")

while(1):
    ret, frame = vcap.read()
    if(type(frame) != type(None)):
    	print("+")             # <= Simulate processing time
    	cv2.imshow('VIDEO', frame)
    	if cv2.waitKey(1) == 27:
        	break
    else:
    	print("-")`

this code is working

amymcgovern, 2018-10-19:
This bug was fixed in the latest update.  The issue is that the new mambo firmware broke ftp which is how we use the groundcamera.  Also, I'm glad your opencv code works but it doesn't in general keep the stream working across all OS's. That's why I did a different approach for vision.  But I am glad it works well for you!

ihsancemil, 2018-10-19:
Vov that was a fast answer :D I will update pyparrot via git then. Thanks... And yes I switch windows to linux to make that code work

closed 2018-10-19 by ihsancemil

ihsancemil, 2018-10-19:
Yes master branch works perfectly