amymcgovern/pyparrot #112: Not able to connect to Bebop 1

derekep, 2018-10-21:
Hey, I’ve been trying to connect to my Bebop 1 drone using this package and I get [WinError 10060] (Timeout Error). I’m connected to the Bebop WiFi and on a Windows computer. I’m not sure what the issue could be but I gave it over 1000 retries in my code. I am running the code from Spyder if there’s any chance that could be problematic. I’m just not sure what the issue could be.

Victor804, 2018-10-21:
Hi,
Can you share your code? To see what the issue is.

derekep, 2018-10-21:
from pyparrot.Bebop import Bebop

bebop = Bebop(drone_type='Bebop')

print("connecting")

success = bebop.connect(20)

print(success)

bebop.set_hull_protection(1)

bebop.set_indoor(1)

bebop.safe_takeoff(10)

bebop.safe_land(10)

bebop.disconnect()

and the error I get says:
TimeoutError: [WinError 10060] A connection attempt failed because the connected party did not properly respond after a period of time, or established connection failed because connected host has failed to respond

amymcgovern, 2018-10-21:
I know what happened, sorry.  I was fixing how the connection is setup for the mambo's new firmware and I forgot that I broke how the other drones work.  parrot broke their firmware *twice* this week and I was teaching a big workshop this week and it was all mambos or I would have noticed earlier.  I'll release a fix in a few minutes. I know exactly what I did (I commented out all my mDNS!).  I need to make it so that it still uses mDNS for bebop but not for mambo (and then hope parrot responds to my request to *fix* their mDNS and FTP services so the mambo goes back to working right!)

amymcgovern, 2018-10-21:
Ok, version 1.5.9 fixes this and I verified that it works on both mambo and bebop 2.  I don't have a bebop 1 but the same bug would have been affecting both and it is now fixed.