amymcgovern/pyparrot #113: Problems of the sensors

lengyuyu2333, 2018-10-23:
Hello everyone.
I met a problem when I try to run a sample, all the instructions are right. But I found:
Success in setting up the wifi network to the drone! 
True
sleeping
 Could not find sensor in list - ignoring for now.  Packet info below. 
(144, 0, 2)
 Could not find sensor in list - ignoring for now.  Packet info below. 
(142, 0, 3)
 Could not find sensor in list - ignoring for now.  Packet info below. 
(134, 0, 3)
 Could not find sensor in list - ignoring for now.  Packet info below. 
(134, 0, 4)
 Could not find sensor in list - ignoring for now.  Packet info below. 
(134, 0, 4)
 Could not find sensor in list - ignoring for now.  Packet info below. 
(134, 0, 4)
 Error: tried to parse a bad sensor packet 
 Could not find sensor in list - ignoring for now.  Packet info below. 
(144, 0, 1)
 Could not find sensor in list - ignoring for now.  Packet info below. 
(144, 0, 1)
 Could not find sensor in list - ignoring for now.  Packet info below. 
(134, 0, 3)
 trying to land 
 Could not find sensor in list - ignoring for now.  Packet info below. 
(144, 0, 1)
 Could not find sensor in list - ignoring for now.  Packet info below. 
(134, 0, 3)
DONE - disconnecting
 disconnecting 
How can I solve these problems of the sensors?

amymcgovern, 2018-10-23:
The sensor missing in the list is because parrot didn't list all of the sensors in their XML file.  I could quit printing out that it is sending that data but I left it in to remind me to add it if they ever add the related XML file to the SDK!

lengyuyu2333, 2019-01-24:
Hello, one more question :)
I am using Parrot Bebop 2
I have used the command bebop.sensors.sensors_dict to show all the return values.
Is there any explains about the values? All in the common.xml?

lengyuyu2333, 2019-01-24:
Oh, I have already found these data in the ardrone3.xml and common.xml, thanks.
By the way, the parrot bebop 2 can only show the changed data but not raw data of the sensors?

amymcgovern, 2019-01-24:
it shows the raw value any time that they are sent.  I didn't hack the firmware so it just parses whatever is sent.

lengyuyu2333, 2019-01-29:
So is there any ways to see the speed(m/s) or spinning speed(rad/s) now?
I just found we can do some limits like maximum and I have not found any about speed yet through the sensors. 
Does the drone has an initial speed?
Thanks for your answers :)

amymcgovern, 2019-01-29:
Did your dad the documentation?  There is extensive documentation on the sensors.  They are all in a dictionary and updated at 10Hz

Bebop sensors:
https://pyparrot.readthedocs.io/en/latest/bebopcommands.html