amymcgovern/pyparrot #114: Flight path generation - droneMapGUI

Litost13, 2018-10-23:
I've been trying to create a trajectory with the parrot mambo using the pyparrot library but I have not got it. Has anyone been able to achieve it? Im trying to recreate the trajectory to do some image processing and land the drone by using openCv library, but the only way out I have found is try to approximate the dron movement on pitch, yaw and roll and generate the "flight path" but its not very useful to me. Maybe I've thinkig if is possible to load the droneMapGUI map into the code in someway or load it into the mambo firmware and generate a kind of mission?.... Has anyone found out how to run the dronMapGUI with the drone?. Thanks.

amymcgovern, 2018-10-24:
mission stuff exists in the parrot SDK I believe.  I won't be releasing it since the GUI was entirely intended for my class and not really for the public release.

machinelearner452, 2018-10-30:
Amy, May I ask if your GUI class is available online?  If so, can you share the details? 
Secondly,  are you aware of good talent with Python or C experience, Drone knowledgable, located on the east coast of the USA that is open to product development targeting a patent application?
As always, thank you for sharing your knowledge and information!

amymcgovern, 2018-10-31:
the droneMapGUI that was referenced above is just a quick script that my AI class uses as part of learning to do intelligent navigation.  It is in the GitHub repo in coursework.