amymcgovern/pyparrot #116: Unable to connect to mambo

mycal-tucker, 2018-11-04:
Hello!

I started playing with PyParrot to try to operate my mambo parrot drone, but I'm having issues connecting over wifi. Here's what I'm doing:

I copied one of the pyparrot demos for streaming back the camera images from the fpv camera into a new PyCharm project on my windows laptop.
I turned on my drone with the camera mounted on top. After initial startup, I see that there's a new wifi network named MAMBO_531746. I connect to that network.
I try to run the demo code but get the following errors:

```
OpenCVAvailable is True
ERROR: ftp login is disabled by parrot firmware 3.0.25 and 26.  Groundcam will not work.
70
C:\Users\mycal\src\drone\stalker-drone\venv\lib\site-packages\pyparrot\images\groundcam.jpg
trying to connect to mambo now
Traceback (most recent call last):
  File "C:/Users/mycal/src/drone/stalker-drone/src/tutorials/mambo_vision_tutorial.py", line 44, in <module>
    success = mambo.connect(num_retries=3)
  File "C:\Users\mycal\src\drone\stalker-drone\venv\lib\site-packages\pyparrot\Minidrone.py", line 391, in connect
    connected = self.drone_connection.connect(num_retries)
  File "C:\Users\mycal\src\drone\stalker-drone\venv\lib\site-packages\pyparrot\networking\wifiConnection.py", line 161, in connect
    handshake = self._handshake(num_retries)
  File "C:\Users\mycal\src\drone\stalker-drone\venv\lib\site-packages\pyparrot\networking\wifiConnection.py", line 305, in _handshake
    tcp_sock.connect(("192.168.99.3", 44444))
TimeoutError: [WinError 10060] A connection attempt failed because the connected party did not properly respond after a period of time, or established connection failed because connected host has failed to respond

Process finished with exit code 1
```

The relevant part of the code is shown below:

```
# you will need to change this to the address of YOUR mambo
#mamboAddr = "e0:14:d0:63:3d:d0"  # From demo
mamboAddr = "9C:B6:D0:88:F5:CD"  # My parrot address.

# make my mambo object
# remember to set True/False for the wifi depending on if you are using the wifi or the BLE to connect
mambo = Mambo(mamboAddr, use_wifi=True)
print("trying to connect to mambo now")
success = mambo.connect(num_retries=3)
print("connected: %s" % success)
```

Do you have any idea why I'm seeing these connection issues? Is it possible I somehow have the wrong MAC address? I found it by looking at network properties after connecting to the MAMBO wifi network.

mycal-tucker, 2018-11-04:
I know this looks like [this issue](https://github.com/amymcgovern/pyparrot/issues/112), but I'm running pyparrot 1.5.10, which is supposed to contain the fix, I believe.

mycal-tucker, 2018-11-04:
Apologies. I didn't mean to close. Just comment.

amymcgovern, 2018-11-04:
The Mac address is not used when you are in wifi mode so that part doesn't matter.  It can be an empty string in wifi mode.  What firmware are you running on your mambo?  Maybe they updated again... They have released multiple updates recently (especially since 3.0.25 broke a bunch of things).  I found 3.0.26 to be quite buggy as well.  I reported the issues and have been hoping for a new release.

mycal-tucker, 2018-11-05:
Thanks for the quick turnaround.

I'm having trouble looking up the firmware version. Do you know how I can find it? I know I've done a reset (by holding the power button for a long time), so at least I don't have any extra modifications floating around.

mycal-tucker, 2018-11-05:
Got it. I connected to the drone through the app. I see the following info:

Software Version: 3.0.18
Hardware Version: HW_02

It looks like this isn't an issue with being a super-recent release. Would you recommend trying to upgrade the firmware?

amymcgovern, 2018-11-05:
No.  Let me check at home with one of my older (not upgraded) drones and make sure I didn't break something that made it not backwards compatible.  I didn't bring a drone with me to work today so I can't check until tonight.  But honestly the new firmware is buggy so I wouldn't upgrade!

mycal-tucker, 2018-11-05:
Ok, I'll hold off. Thanks.

amymcgovern, 2018-11-06:
Ok, I have one at home that is running 3.0.17 and it connected just fine.   Wrote a sample connection program shown below.  It did give me that error right when I first connected to the drone but I just ran it again and it worked fine.  Sometimes it takes a few seconds to hand shake to the drone's wifi.

```python
from pyparrot.Minidrone import Mambo

mambo = Mambo(use_wifi=True)

success = mambo.connect(num_retries=3)

print(success)

mambo.smart_sleep(3)

mambo.disconnect()
```

mycal-tucker, 2018-11-06:
Great. Thanks for the help debugging. I'll test some more tomorrow morning
and post my results.

mycal-tucker, 2018-11-06:
Actually, it looks like I can't even connect to my camera using the phone app. That tells me that this isn't an issue with PyParrot.

I have contacted Parrot to try to get support from them.

closed 2018-11-06 by mycal-tucker