amymcgovern/pyparrot #117: Could not find codec parameters for stream 0 (Video: h264, none): unspecified size

stevenhedges, 2018-11-12:
Hi

I have recently dusted off an old Bebop 1 drone, intending to use it for a deep learning computer vision project.

However, I've stumbled at the first hurdle as I cannot get either demoBebopVision.py or demoBebopVisionGUI.py to work.

I am running on Ubuntu 18.04 and indoors only (at the moment).  When running the first programme I get the console output below this message.

Is there any more info needed to help resolve this issue?  I have tried playing with 'analyzeduration' and 'probesize' options as suggested, but  with no success.

Thanks in advance

Steven

Opening ffmpeg
ffmpeg -protocol_whitelist "file,rtp,udp" -i /home/steven/drone/pyparrot/pyparrot/utils/bebop.sdp -r 30 image_%03d.png &
starting vision thread
Opening non-blocking readers

ffmpeg version 3.4.4-0ubuntu0.18.04.1 Copyright (c) 2000-2018 the FFmpeg developers

   built with gcc 7 (Ubuntu 7.3.0-16ubuntu3)

  configuration: --prefix=/usr --extra-version=0ubuntu0.18.04.1 --toolchain=hardened --libdir=/usr/lib/x86_64-linux-gnu --incdir=/usr/include/x86_64-linux-gnu --enable-gpl --disable-stripping --enable-avresample --enable-avisynth --enable-gnutls --enable-ladspa --enable-libass --enable-libbluray --enable-libbs2b --enable-libcaca --enable-libcdio --enable-libflite --enable-libfontconfig --enable-libfreetype --enable-libfribidi --enable-libgme --enable-libgsm --enable-libmp3lame --enable-libmysofa --enable-libopenjpeg --enable-libopenmpt --enable-libopus --enable-libpulse --enable-librubberband --enable-librsvg --enable-libshine --enable-libsnappy --enable-libsoxr --enable-libspeex --enable-libssh --enable-libtheora --enable-libtwolame --enable-libvorbis --enable-libvpx --enable-libwavpack --enable-libwebp --enable-libx265 --enable-libxml2 --enable-libxvid --enable-libzmq --enable-libzvbi --enable-omx --enable-openal --enable-opengl --enable-sdl2 --enable-libdc1394 --enable-libdrm --enable-libiec61883 --enable-chromaprint --enable-frei0r --enable-libopencv --enable-libx264 --enable-shared

  libavutil      55. 78.100 / 55. 78.100
  libavcodec     57.107.100 / 57.107.100
  libavformat    57. 83.100 / 57. 83.100
  libavdevice    57. 10.100 / 57. 10.100
  libavfilter     6.107.100 /  6.107.100
  libavresample   3.  7.  0 /  3.  7.  0
  libswscale      4.  8.100 /  4.  8.100
  libswresample   2.  9.100 /  2.  9.100
  libpostproc    54.  7.100 / 54.  7.100

[sdp @ 0x55e079854460] Could not find codec parameters for stream 0 (Video: h264, none): unspecified size

Consider increasing the value for the 'analyzeduration' and 'probesize' options

Input #0, sdp, from '/home/steven/drone/pyparrot/pyparrot/utils/bebop.sdp':

  Duration: N/A, bitrate: N/A

    Stream #0:0: Video: h264, none, 90k tbr, 90k tbn, 180k tbc

Output #0, image2, to 'image_%03d.png':

Output file #0 does not contain any stream

Having trouble connecting to the camera 1.  A reboot of the mambo may help.

amymcgovern, 2018-11-15:
Sorry to get behind. This is pretty odd.  Did you try it more than once?  I don't have a bebop 1 but the times I have had access to one, the vision stream has worked fine.  I have a bebop 2 and it isn't having issues.

stevenhedges, 2018-11-15:
Hi Amy

I think that the problem might be in the bebop drone rather than pyparrot. 

I installed your code onto a Mac laptop and had the same error accessing the video stream. 

In both cases I can control the flight of the drone, so the code is actually connecting and interacting correctly. 

Unfortunately not being able to access the video stream rather stymies my plan to teach the drone to play hide and seek!

Steven

PS I’m not sure it is relevant but I do get a lot of messages about the need to handle low latency data.

Maryamr314, 2020-05-17:
I have the same problem with Anafi4k. I can't see the video. @stevenhedges did you solve the problem?