amymcgovern/pyparrot #118: demoMamboVision.py does not work!

YN312, 2018-11-15:
I ran python demoMamboVision.py. But...

OpenCVAvailable is True
ERROR: ftp login is disabled by parrot firmware 3.0.25 and 26.  Groundcam will not work.
61
/Users/Nagaccho/anaconda/lib/python3.6/site-packages/pyparrot/images/groundcam.jpg
trying to connect to mambo now
{"d2c_port": 43210, "controller_type": "computer", "controller_name": "pyparrot"}
{'status': 0, 'c2d_port': 6000, 'c2d_update_port': 51, 'c2d_user_port': 21}
c2d_port is 6000
starting listening at 
 Success in setting up the wifi network to the drone! 
connected: True
sleeping
 Could not find sensor in list - ignoring for now.  Packet info below. 
(2, 25, 3)
 Could not find sensor in list - ignoring for now.  Packet info below. 
(2, 25, 5)
Preparing to open vision
61
/Users/Nagaccho/anaconda/lib/python3.6/site-packages/pyparrot/images
/Users/Nagaccho/anaconda/lib/python3.6/site-packages/pyparrot/utils
Opening ffmpeg
starting vision thread
Opening non-blocking readers
in save pictures on image 0 
ffmpeg version 4.1 Copyright (c) 2000-2018 the FFmpeg developers

  built with Apple LLVM version 10.0.0 (clang-1000.11.45.5)

  configuration: --prefix=/usr/local/Cellar/ffmpeg/4.1 --enable-shared --enable-pthreads --enable-version3 --enable-hardcoded-tables --enable-avresample --cc=clang --host-cflags= --host-ldflags= --enable-ffplay --enable-gpl --enable-libmp3lame --enable-libopus --enable-libsnappy --enable-libtheora --enable-libvorbis --enable-libvpx --enable-libx264 --enable-libx265 --enable-libxvid --enable-lzma --enable-opencl --enable-videotoolbox

  libavutil      56. 22.100 / 56. 22.100

  libavcodec     58. 35.100 / 58. 35.100

  libavformat    58. 20.100 / 58. 20.100

  libavdevice    58.  5.100 / 58.  5.100

  libavfilter     7. 40.101 /  7. 40.101

  libavresample   4.  0.  0 /  4.  0.  0

  libswscale      5.  3.100 /  5.  3.100

  libswresample   3.  3.100 /  3.  3.100

  libpostproc    55.  3.100 / 55.  3.100

[h264 @ 0x7fa7dd804000] cabac_init_idc 12 overflow

[h264 @ 0x7fa7dd804000] decode_slice_header error

[h264 @ 0x7fa7dd804000] cabac_init_idc 14 overflow

[h264 @ 0x7fa7dd804000] decode_slice_header error

[h264 @ 0x7fa7dd804000] cabac_init_idc 14 overflow

[h264 @ 0x7fa7dd804000] decode_slice_header error

[h264 @ 0x7fa7dd804000] cabac_init_idc 14 overflow

[h264 @ 0x7fa7dd804000] decode_slice_header error

[h264 @ 0x7fa7dd804000] cabac_init_idc 14 overflow

[h264 @ 0x7fa7dd804000] decode_slice_header error

[... 862 lines cut ...]
	21  libpython3.6m.dylib                 0x000000010153b992 _PyEval_EvalFrameDefault + 27346
	22  libpython3.6m.dylib                 0x000000010154067c _PyFunction_FastCallDict + 348
	23  libpython3.6m.dylib                 0x0000000101474cf7 _PyObject_FastCallDict + 247
	24  libpython3.6m.dylib                 0x0000000101474e15 _PyObject_Call_Prepend + 149
	25  libpython3.6m.dylib                 0x0000000101474b30 PyObject_Call + 96
	26  libpython3.6m.dylib                 0x000000010153bc75 _PyEval_EvalFrameDefault + 28085
	27  libpython3.6m.dylib                 0x000000010154045d fast_function + 381
	28  libpython3.6m.dylib                 0x000000010153f1c1 call_function + 401
	29  libpython3.6m.dylib                 0x000000010153b992 _PyEval_EvalFrameDefault + 27346
	30  libpython3.6m.dylib                 0x000000010154045d fast_function + 381
	31  libpython3.6m.dylib                 0x000000010153f1c1 call_function + 401
	32  libpython3.6m.dylib                 0x000000010153b992 _PyEval_EvalFrameDefault + 27346
	33  libpython3.6m.dylib                 0x000000010154067c _PyFunction_FastCallDict + 348
	34  libpython3.6m.dylib                 0x0000000101474cf7 _PyObject_FastCallDict + 247
	35  libpython3.6m.dylib                 0x0000000101474e15 _PyObject_Call_Prepend + 149
	36  libpython3.6m.dylib                 0x0000000101474b30 PyObject_Call + 96
	37  libpython3.6m.dylib                 0x0000000101585996 t_bootstrap + 70
	38  libsystem_pthread.dylib             0x00007fff5c39133d _pthread_body + 126
	39  libsystem_pthread.dylib             0x00007fff5c3942a7 _pthread_start + 70
	40  libsystem_pthread.dylib             0x00007fff5c390425 thread_start + 13
)
libc++abi.dylib: terminating with uncaught exception of type NSException
Abort trap: 6

The video does not start. Help me!

YN312, 2018-11-16:
The OS version is
macOS Mojave version 10.14.

amymcgovern, 2018-12-18:
That's what I have and it is working fine.  I'm going to assume a library conflict in your install somewhere?  Maybe try a reinstall?

closed 2019-01-09 by amymcgovern