amymcgovern/pyparrot #121: mamboVision.closeVideo does not work as expected in demoMamboVision

piyushnariani, 2018-11-21:
I have been trying to take the picture using the FPV camera mounted on Parrot mambo. The expected behavior is that it should click x pictures per second (frame rate-buffer size) and once closeVideo function is called, it should stop taking those pictures. However, in my case, it continuously takes pictures and keeps storing pictures even after mambo.disconnect is called. I have to manually disconnect from Mambo Wifi to stop it from flooding the directory with pictures.

amymcgovern, 2018-11-21:
are you using the ffmpeg or the libVLC approach?  It sounds like your subprocess isn't being killed.

amymcgovern, 2018-11-22:
What OS?  I can for sure say it isn't killing the ffmpeg subprocess based on your description.

piyushnariani, 2018-11-22:
Windows 10

piyushnariani, 2018-11-23:
OpenCVAvailable is True
ERROR: ftp login is disabled by parrot firmware 3.0.25 and 26.  Groundcam will not work.
29
D:\Practice\pyparrot\pyparrot\images\groundcam.jpg
trying to connect to mambo now
{"d2c_port": 43210, "controller_type": "computer", "controller_name": "pyparrot"}
{'status': 0, 'c2d_port': 6000, 'c2d_update_port': 51, 'c2d_user_port': 21}
c2d_port is 6000
starting listening at  Success in setting up the wifi network to the drone! 
connected: True
sleeping

 Could not find sensor in list - ignoring for now.  Packet info below. 
(2, 25, 3)
 Could not find sensor in list - ignoring for now.  Packet info below. 
(2, 25, 5)
taking off!
flying state is hovering
Flying direct: going up
Preparing to open vision
29
D:\Practice\pyparrot\pyparrot\images
D:\Practice\pyparrot\pyparrot\utils
Opening ffmpeg
starting vision thread
Opening non-blocking readers
ffmpeg version N-92498-g77bf85515e Copyright (c) 2000-2018 the FFmpeg developers

  built with gcc 8.2.1 (GCC) 20181017

  configuration: --enable-gpl --enable-version3 --enable-sdl2 --enable-fontconfig --enable-gnutls --enable-iconv --enable-libass --enable-libbluray --enable-libfreetype --enable-libmp3lame --enable-libopencore-amrnb --enable-libopencore-amrwb --enable-libopenjpeg --enable-libopus --enable-libshine --enable-libsnappy --enable-libsoxr --enable-libtheora --enable-libtwolame --enable-libvpx --enable-libwavpack --enable-libwebp --enable-libx264 --enable-libx265 --enable-libxml2 --enable-libzimg --enable-lzma --enable-zlib --enable-gmp --enable-libvidstab --enable-libvorbis --enable-libvo-amrwbenc --enable-libmysofa --enable-libspeex --enable-libxvid --enable-libaom --enable-libmfx --enable-amf --enable-ffnvcodec --enable-cuvid --enable-d3d11va --enable-nvenc --enable-nvdec --enable-dxva2 --enable-avisynth

  libavutil      56. 23.101 / 56. 23.101

  libavcodec     58. 39.100 / 58. 39.100

  libavformat    58. 22.100 / 58. 22.100

  libavdevice    58.  6.100 / 58.  6.100

  libavfilter     7. 46.100 /  7. 46.100

  libswscale      5.  4.100 /  5.  4.100

  libswresample   3.  4.100 /  3.  4.100

  libpostproc    55.  4.100 / 55.  4.100

[udp @ 000001fc2a5aed00] 'circular_buffer_size' option was set but it is not supported on this build (pthread support is required)

[udp @ 000001fc2a5bf000] 'circular_buffer_size' option was set but it is not supported on this build (pthread support is required)

[h264 @ 000001fc2a5ae580] cabac_init_idc 6 overflow

[h264 @ 000001fc2a5ae580] decode_slice_header error

[h264 @ 000001fc2a5ae580] cabac_init_idc 8 overflow

[h264 @ 000001fc2a5ae580] decode_slice_header error

[h264 @ 000001fc2a5ae580] cabac_init_idc 8 overflow

[h264 @ 000001fc2a5ae580] decode_slice_header error

[h264 @ 000001fc2a5ae580] cabac_init_idc 6 overflow

[h264 @ 000001fc2a5ae580] decode_slice_header error

[h264 @ 000001fc2a5ae580] cabac_init_idc 6 overflow

[... 796 lines cut ...]
Input #0, rtsp, from 'rtsp://192.168.99.1/media/stream2':

  Metadata:

    title           : media/stream2

    comment         : A Seminar on the session description protocol

  Duration: N/A, start: 0.032000, bitrate: N/A

    Stream #0:0: Video: h264 (High), yuv420p(progressive), 640x360, 30 tbr, 90k tbn, 180k tbc

Stream mapping:

  Stream #0:0 -> #0:0 (h264 (native) -> png (native))

Success in opening vision is True
Vision successfully started!
landing
flying state is hovering
 trying to land 
disconnecting
 disconnecting 

Process finished with exit code 0