amymcgovern/pyparrot #123: Accessing Sensor data

derekep, 2018-12-01:
I've been trying to determine if, with your package, you are able to access actual readings from the sensors or not and I have been unable to figure it out. In the freeflight pro app that parrot developed, they have a reading of the height of the drone above the ground, and I would like to be able to access the height through python for a project I am working on. If this is feasible with your package, please let me know.

amymcgovern, 2018-12-01:
Mambo or bebop?  Answer is yes either way 

https://pyparrot.readthedocs.io/en/latest/minidronecommands.html

https://pyparrot.readthedocs.io/en/latest/bebopcommands.html#bebop-sensors

derekep, 2018-12-01:
bebop

derekep, 2018-12-01:
I've seen those commands, but was unable to work out how to use them. The documentation has me confused. Is there an example script where this data is called that I can look at?

amymcgovern, 2018-12-01:
Print out the contents of bebop.sensors.sensors_dict and you will every sensor that is sent.  I could write up an example using the sensors but that is where they all are.  The special ones (like battery) are stored separately but everything else is in the dictionary.  battery is bebop.sensors.battery and the flying state is bebop.sensors.flying_state.

derekep, 2018-12-05:
Wow, that was super simple. I should have been able to figure that out on my own, but thanks for helping me out! That does exactly what I want. Thanks.

closed 2019-01-09 by amymcgovern