amymcgovern/pyparrot #124: Mambo sensors_dict DronePosition_posx

marcusabate, 2018-12-02:
Hello,

the ``Minidrone.MinidroneSensors`` class gives us velocities but I've also noticed ``DronePosition_posx``, ``_posy`` and ``_posz`` in the sensors_dict. How are these estimates done? Is it just dvdt or is there some more sensor fusion happening with the accelerometer etc behind the scenes? 
I ask because I'd like to develop a precise position controller, so position estimates are important.

amymcgovern, 2018-12-03:
That is a new sensor with the new firmware!  Hooray for parrot for finally sending that data.  It was in the xml file so it has the following documentation:

Get the drone position from takeoff point (0, 0, 0, 0).
			The orthonormal basis is fixed at this point. The axis are
			oriented the following way :
			* X : From the rear of the drone to its front.
			* Y : From the right of the drone to its left.
			* Z : Orthogonal to X and Y and oriented upward.
			* Psi : From 180 to -180 in the clockwise direction,

I will update pyparrot to also make it more accessible but you can easily use it from the sensors_dict, which was intended as a catch all for any new or extra sensors anyway.

amymcgovern, 2018-12-04:
I did some testing tonight and it seems that the data is somewhat time lagged.  I did a quick program where I took off, flew forward, and watched the posx variable.  it seems like it is correct but very time lagged.  I will post about it on the dev forum for parrot and see what they say the lag is.

marcusabate, 2018-12-04:
I've seen the same thing in some of my tests.
The documentation says that the WiFi sensor update rate is 2Hz, however I've been timing between sensor callbacks and it seems like it goes way faster than that. Do you know if the new firmware might be responsible for this, or if it's more likely I'm doing timing wrong in some way? I can't find any references on the developer forum.

amymcgovern, 2018-12-04:
There are a bunch of posts on there from me asking and them confirming the 2Hz rate but those are all old.  I'm still seeing about 500ms between sensor updates though.  Also, the drone position ts variable is never updated (it is always 0).  So that doesn't help anyway since you can't check the timestamp!  But the ts on speed is still at 500 ms.

marcusabate, 2018-12-04:
Hmm I was just using ``time.perf_counter()`` to check sensor update dt. When I do that, many of the updates seem to happen on the order of 0.2ms.
I've just done those tests tracking ``speed_ts`` as well and a lot of the updates show something to the effect of:
```
mambo.sensors.speed_ts - self.time_since_last_update = 0
```
Maybe a rounding problem since the ts comes in as an int? 
When the updates do take time, I am seeing the occasional 500ms dt like you've seen. This happens rarely and typically during the beginning of my flight (around takeoff).
This is all happening in a sensor callback function. I basically save the data to a list and then I view it post flight. The dt I measure using both ``time.perf_counter()`` and ``mambo.sensors.speed_ts`` is done at the top of every callback.
Also, this last round of test was conducted over BLE but I see the same results over WiFi.
Thoughts?

marcusabate, 2018-12-04:
I'm running 3.0.26. I think I've found my mistake though.

My data shows the callback is *called* every 0.2ms. The docs suggest that the callback is called anytime **a** sensor is updated but I'm not sure whether that means any sensor update or a full sensor suite update. Regardless, in my data file I have looked for instances of the same position value and see that it persists for 10 callback method calls. Add the dt's of all 10 of those calls and I get 500ms before a new position value pops up.

So turns out the data does update at 2hz, but the callback seems to be called way more often than that. Seems to be pretty regularly 10 calls between updates for position. For velocity it is sometimes 8, sometimes 2, etc.

amymcgovern, 2018-12-04:
I was just about to suggest that was the issue.  It is called anytime *any* sensor is updated.  And I think some are coming in more frequently.  I don't have any heartbeats inside that code.

closed 2019-01-09 by amymcgovern