amymcgovern/pyparrot #126: Cannot connect to parrot Mambo using BLE, but can find it

BolunDai0216, 2018-12-04:
I can find the parrot mambo drone using findMiniDrone.py and it gives me an address, I added that address to the demoMamboTricks.py script, and ran it, then it shows me:

OpenCVAvailable is False
trying to connect
trying to connect to the minidrone at address D0:3A:C0:0C:E6:21
retrying connections
trying to connect to the minidrone at address D0:3A:C0:0C:E6:21
retrying connections
trying to connect to the minidrone at address D0:3A:C0:0C:E6:21
retrying connections
trying to connect to the minidrone at address D0:3A:C0:0C:E6:21
retrying connections
trying to connect to the minidrone at address D0:3A:C0:0C:E6:21
retrying connections
trying to connect to the minidrone at address D0:3A:C0:0C:E6:21
retrying connections
trying to connect to the minidrone at address D0:3A:C0:0C:E6:21
retrying connections
trying to connect to the minidrone at address D0:3A:C0:0C:E6:21
retrying connections
trying to connect to the minidrone at address D0:3A:C0:0C:E6:21
retrying connections
connected: False

Can anyone help me with this issue (I'm using BLE instead of wifi).

amymcgovern, 2018-12-04:
that is odd.  I assume your phone has no issues connecting to it over BLE?

BolunDai0216, 2018-12-04:
I am connecting to it using a Linux Desktop running ubuntu

BolunDai0216, 2018-12-04:
And yes my phone has no issue connecting to mambo using BLE, and running the official Simulink library over bluetooth also has no issues.

amymcgovern, 2018-12-05:
I'll test things tonight on my linux box and see if there has been an accidental bug introduced when I was trying to keep up with all the firmware changes from parrot.  Can you confirm your firmware version?

BolunDai0216, 2018-12-05:
We got it working, I am not sure what the current version is, but when we were trying with 3.0.26 (the newest firmware) it just could not connect, so we chose an older version.

amymcgovern, 2018-12-05:
Excellent.  I will make sure that the newest firmware is working again then.  rI'm sure it has to do with how the handshake changed in the latest firmware.  I have several with older firmware and several with newer so I'll ensure it is working on both.

ParisaPouya, 2019-02-05:
Hi, 

I am facing the same issue while trying to connect to mambo parrot using BLE. I am able to see the device using BLE and I am able to fly it using the app. However, I am not sure why it is failing to connect when I run the examples. 
By the way, I don't know what firmware version I am using. 

I would appreciate any help.

ParisaPouya, 2019-02-07:
Hi, 

Thank you for your fast response. I was able to resolve the issue. It looks like when we try to run one of the example codes and connect via bluetooth for the first time, we should initially run the code, let the code wait for connection, and then try actually connecting to the device using the bluetooth. I tried connecting this way and it works!

bathonSpidey, 2020-02-29:
Is there a way to get the BLE address on windows 10? I have tried emulating using VM ware raspberry buster and I get raise BTLEManagementError failed to execute 'le on'

amymcgovern, 2020-02-29:
the BLE library I used is only compatible with linux

bathonSpidey, 2020-03-01:
Yes, I got an error stating "ERROR: you are trying to use a BLE connection on a system that doesn't have BLE installed. ". I used VM ware with Ubuntu 18.04. I am assuming it is due to the fact that my laptop does not have a BLE.