amymcgovern/pyparrot #128: Does Mambo drone send the 'Flying' Piloting State?

anoop4, 2018-12-14:
I see that the Parrot mambo drone sends all the different piloting state except for the 'Flying' state.

We have to currently reply on smart_sleep() to prevent sequential fly_direct() commands from being skipped.

I was wondering if the PilotingState change from 'hovering' to 'flying' can be used as an indication of a previous move complete. This can help in avoiding the samrt_sleep() commands in the pyparrot examples.

Unfortunately, I don't see the 'Flying' state sent by the drone with the latest 3.0.26 FW.

Any thoughts?

amymcgovern, 2018-12-14:
are you using wifi or BLE?

anoop4, 2018-12-14:
Wifi

amymcgovern, 2018-12-14:
windows?  I've seen an issue with windows sensors specifically:

https://pyparrot.readthedocs.io/en/latest/faq.html#i-m-using-windows-and-my-drone-gives-me-lots-of-timeout-errors

anoop4, 2018-12-14:
Yes I am seeing this in my windows system. Will get back to you after checking in my Mac system.

anoop4, 2018-12-18:
I am getting the same behavior in my mac system as well. The drone doesn't seem to send the 'Flying' state at all. 

I get other states such as taking off, hovering, landing, landed, initializing, and emergency

closed 2019-01-09 by amymcgovern