amymcgovern/pyparrot #129: Parrot swing not connecting to windows pc bluetooth

ElLobo007, 2018-12-17:
Hi, I am using a pc with Microsoft 10, and I am trying to connect to my Parrot Swing. When I connect to the swing via bluetooth, green light on the drones becomes steady green and bluetooth says connected. But a few seconds later, it disconnects. 
I don't know whether to run the code on Anaconda cmd prompt, python, or Visual Studio since I am using Windows?
My goal is to connect my pc to Parrot swing and control and monitor its state via my pc.
It connects to phone and flypad normally
Any help is much appreciated

amymcgovern, 2018-12-19:
That is odd.  I don't have a swing (the swing part was contributed by another user) but I wonder if the new firmware is messing with it.  does it have any errors?

ElLobo007, 2018-12-19:
No errors as far as I can see, and since it connects to the flypad and phone without fault. I've been using the pyparrot Documentation to try to conect my pc to the drone, and I've been able to install everything including the parrot libraries, except for the BLE, which has been giving me so many errors in the terminal (Is it necessary to install BLE if my PC already has bluetooth?). 
On pg 6 of the pyparrot Docs, it has this code for BLE connection "sudo findMinidrone", what is the Microsoft equivalent of this.

Thanks

ElLobo007, 2018-12-20:
Yeah, I had the same thought. I think you're right, I'll have a look for a good emulator, and I'll post the feedback