amymcgovern/pyparrot #131: demoMamboGroundcam.py throws this error

piyushnariani, 2018-12-18:
Hello, I have been trying to use groundcam to take some pictures, however, I am getting this error.

_Traceback (most recent call last):
  File "~/pyparrot/examples/demoMamboGroundcam.py", line 25, in <module>
    pic_success = mambo.take_picture()
  File "~\pyparrot\pyparrot\Minidrone.py", line 551, in take_picture
    list = self.groundcam.get_groundcam_pictures_names()
  File "~\pyparrot\pyparrot\Minidrone.py", line 268, in get_groundcam_pictures_names
    return list()
UnboundLocalError: local variable 'list' referenced before assignment_

I appreciate your guidance regarding the same.

amymcgovern, 2018-12-18:
Parrot broke the interface to the ground camera in the latest firmware.  What firmware are you running?  It should still work if you are several versions old.  I can debug on my end if I know what firmware you have.  Thanks.

piyushnariani, 2018-12-27:
How do I find the firmware version of the parrot mambo drone?

amymcgovern, 2018-12-27:
Your mobile app (phone) will tell you.

piyushnariani, 2018-12-28:
Its 3.0.25

amymcgovern, 2018-12-28:
Ok that is the version where the groundcamera is completely broken.  If you can downgrade a version, that would be best.  3.0.26 still has issues also.

piyushnariani, 2018-12-28:
Do we have any manual available to downgrade the version? and which version should I downgrade it to?

CofeDrink68, 2019-01-17:
How can I downgrade? Because I don't see anything in the app for version management

CofeDrink68, 2019-03-10:
I tried to find the previous mambo version (version 3.0.24) but this didn't work does someone have an idea to find this version and how to upload it to the mambo ?
I'm working on a school project for augmented reality and I would finish as quickly as possible.

Anyway, thank you for your library, it's really helpful.

amymcgovern, 2019-03-10:
Parrot took all their old firmware off their webpage, alas.  What is your error?  The library should be working just fine with the current firmware (except for the downward camera issue, which I cannot fix due to parrot breaking it)

CofeDrink68, 2019-03-10:
Exactly like piyushnariani, the file demoMomboGroundcam.py has thrown this error : 

Traceback (most recent call last):
File "/pyparrot/examples/demoMamboGroundcam.py", line 25, in
pic_success = mambo.take_picture()
File "\pyparrot\pyparrot\Minidrone.py", line 551, in take_picture
list = self.groundcam.get_groundcam_pictures_names()
File "\pyparrot\pyparrot\Minidrone.py", line 268, in get_groundcam_pictures_names
return list()
UnboundLocalError: local variable 'list' referenced before assignment

But this error was at the end of 2018. So I searched and I found this issue and I begin searching for an older firmware to apply to the mambo

amymcgovern, 2019-03-10:
Are you using the latest pyparrot?  I don't see that error with the current parrot firmware and the pyparrot library.

featureg, 2019-03-20:
I have the same issue with firmware 3.0.26.

limsm3, 2019-08-19:
It could be simply solved by changing the every assignments of list except for  'function'  list() into some other variable.  You will also have to uncomment the part in Minidrone.py where it enables the ftp connection with the drone.