amymcgovern/pyparrot #132: How to realtime bebop 2 with opencv ?

ThorPham, 2018-12-31:
I want to use bebop realtime to face recognition. But i don't know how to take data realtime with open cv?
Can you help me?

amymcgovern, 2018-12-31:
https://pyparrot.readthedocs.io/en/latest/vision.html

ThorPham, 2019-01-01:
Thank you, Happy New Year @amymcgovern .
I can show GUI of bebop 2 to realtime. But can you so me function(or variable..)or whatever can i get directly video. Because I need to put it to OPENCV function  to read and processing  for face recognition

amymcgovern, 2019-01-09:
As has been mentioned in multiple issue threads, there is a big in opencv that prevents the streaming from working reliably on all OS's.  there is apparently a solution for windows but I haven't added it since it is windows specific and I've tried to ensure things work across OSes.

alishibli97, 2019-01-09:
Can you share the solution on windows please ? I've been trying for weeks
I'm trying to make it work on windows for now
Thanks in advance.

amymcgovern, 2019-01-09:
https://github.com/amymcgovern/pyparrot/issues/102

malfonso0, 2019-05-08:
hi.. see the updated answerd in
https://github.com/amymcgovern/pyparrot/issues/102#issuecomment-487587094
regards