amymcgovern/pyparrot #135: Mambo random disconnections using BLE

revanth7667, 2019-01-08:
I'm trying to control the mambo(fly) by  keyboard commands using BLE connection.
I'm experiencing random disconnections from the mambo when i enter  the commands , it is not due to the Bluetooth range as i checked the signal strength and the demo codes demoMamboTricks and demoMamboDirectFlight work without any issues.
The nature of the disconnects is very random sometimes i do not get any and sometimes it happens in the first command itself or in later commands.
i tried changing the mambo.smart_sleep() to different values (0,2,5) but it did not make any difference

System:
**Ubuntu 18.04 LTS**
and the drone is running the latest firmware **3.0.26**

I'm attaching the code and the output for reference
**Code:**
```python
# code for direct control of drone

from pyparrot.Minidrone import Mambo

mamboAddr = "d0:3a:81:ae:e6:24"  # address of mambo Minidrone

mambo = Mambo(mamboAddr, use_wifi=False)  # creating mambo object

# connecting to drone, number of tries =3
print("trying to connect")
success = mambo.connect(num_retries=3)
print("connected: %s" % success)

if success:
    print("sleeping")
    mambo.smart_sleep(2)
    mambo.ask_for_state_update()
    mambo.smart_sleep(2)

    print("taking off!")  # taking off
    mambo.safe_takeoff(5)

    while True:
        command = ''
        try:
            command = input("enter the direction or press e to exit: ")
            command = command.lower()
        except EOFError:
            print("EOFError occurred")
            break
        if command == 'f':  # forward direction
            print("going forward")
            mambo.fly_direct(roll=0, pitch=50, yaw=0, vertical_movement=0, duration=0.5)
            #mambo.smart_sleep(5)

        elif command == 'b':  # backward direction
            print("going backwards")
            mambo.fly_direct(roll=0, pitch=-50, yaw=0, vertical_movement=0, duration=0.5)
            #mambo.smart_sleep(5)

        elif command == 'l':  # left direction
            print("Turning left")
            mambo.turn_degrees(-90)
            #mambo.smart_sleep(5)

        elif command == "r":  # right direction
            print("Turning right ")
            mambo.turn_degrees(90)
            #mambo.smart_sleep(5)

        elif command == "e":  # exiting after landing
            print("landing")
            mambo.safe_land(5)
            mambo.smart_sleep(5)
            break

        else:
            print("Please enter valid command")

    print("disconnect")
    mambo.disconnect()
else:
    print("""connection failure!
 please check device status and try again""")

print("Thank you!")

```

**Output:**

OpenCVAvailable is False
trying to connect
 trying to connect to the minidrone at address d0:3a:81:ae:e6:24 
 connected!  Asking for services and characteristics 
 magic handshake to make the drone listen to our commandsandsensors 
 initializing notification delegate 
connected: True
sleeping
 Could not find sensor in list - ignoring for now.  Packet info below. 
(2, 25, 3)
 Could not find sensor in list - ignoring for now.  Packet info below. 
(2, 25, 5)
taking off!
enter the direction or press e to exit: l
Turning left
enter the direction or press e to exit: r
Turning right 
 reconnecting to send packet 
 trying to re-connect to the minidrone at address d0:3a:81:ae:e6:24 
 connected!  Asking for services and characteristics 
 magic handshake to make the drone listen to our commandsandsensors 

after the last line the code stops executing and does not work further, in this particular case the drone did not turn right  so i guess the connection was interrupted before the command was sent

revanth7667, 2019-01-17:
Update:
 the direct keyboard control is still not working, however i tried to give all the desired actions using a single string at the beginning and this works without any issues so far, however this is still not useful for the project I'm trying to implement 
I'm posting the code for reference:
```python
# control using single input sequence

from pyparrot.Minidrone import Mambo

mamboAddr = "d0:3a:81:ae:e6:24"  # address of mambo Minidrone

mambo = Mambo(mamboAddr, use_wifi=False)  # creating mambo object

def forward():
    print("going forward")
    mambo.fly_direct(roll=0, pitch=50, yaw=0, vertical_movement=0, duration=0.5)
    return

def backward():
    print("going backwards")
    mambo.fly_direct(roll=0, pitch=-50, yaw=0, vertical_movement=0, duration=0.5)
    return

def left():
    print("Turning left")
    mambo.turn_degrees(-90)
    return

def right():
    print("Turning right")
    mambo.turn_degrees(90)
    return

command = ''
try:
    command = input("enter the direction or press e to exit: ")
    command = command.lower()
except EOFError:
    print("EOFError occurred")

print("trying to connect")
success = mambo.connect(num_retries=3)
print("connected: %s" % success)

if success:

    print("taking off!")  # taking off
    mambo.safe_takeoff(5)
    mambo.smart_sleep(2)

    for i in range(0, len(command)):
        mambo.smart_sleep(2)
        x = command[i]

        if x == 'f':
            forward()
            continue
        elif x == 'b':
            backward()
            continue
        elif x == 'l':
            left()
            continue
        elif x == 'r':
            right()
            continue
        elif x == 'e':
            print("landing")
            mambo.safe_land(5)
            mambo.smart_sleep(5)
            break
        else:
            continue
    else:
        print("landing")
        mambo.safe_land(5)
        mambo.smart_sleep(5)

    print("disconnect")
    mambo.disconnect()
else:
    print("""connection failure!
 please check device status and try again""")

print("Thank you!")

```
I'll try to implement using pygame and update if i find any solution or work around

amymcgovern, 2019-01-17:
I have found that the mambo likes to disconnect if you wait too long for keyboard input.  I don't know the solution other than essentially what you did.  I'd do a try and then except would reconnect.

Victor804, 2019-01-19:
Hi, 
I think you could use the keyboard events with pygame.
[https://stackoverflow.com/questions/16044229/how-to-get-keyboard-input-in-pygame](https://stackoverflow.com/questions/16044229/how-to-get-keyboard-input-in-pygame)