amymcgovern/pyparrot #141: Different mambos with different alignment problems

walber000, 2019-02-05:
Hello Amy,

I have two parrot mambo's, and I have problems with both of them:
First mambo: when I apply a pitch command, the movement tends to the left
Second mambo, takeoff tends a lot to the right (usually almost one meter, then the drone try to correct itself going back to the inicial position), and its movement tends to the right when applying pitch command
Both of them have the same firmware (3.0.26)
Is this a hardware issue? I tried "correcting" it by applying a roll command together with pitch, but I thing it's not a very good solution.
(Another information: when applying very small pitch, the foward movement is slowed, but the "bias" movement - left or right - keeps the same speed)
(When applying all angles zero, the drone stops moving, there is no side movements. When applying only yaw or vertical movement, there is no problems as well. When applying roll commands, it can be seen that one direction is "stronger" than the other, with the same angle)

Regards,
Walber

amymcgovern, 2019-02-06:
This is a hardware issue and not software.  Likely you have some mildly bent blades.  Mambos are quite robust to crashes and bent blades but they lose their precision flying over time as those blades wear.  You can try replacing the blades with fresh ones (there are cheap off-brand ones online as well as the regular parrot ones).  I go through a lot of mambo blades since I have students programming on them all the time!

walber000, 2019-02-06:
I tried switching the blades of each drone between themselves, expecting that the behavior of both would be switched as well, but nothing changed. I tried mixing the blades as well (one drone with two blades of itself and two blades of the other drone, always respecting the blades orientation/curvature).
I will buy new blades as soon as possible and try them.

amymcgovern, 2019-02-06:
Interesting.  If you swapped blades, I would definitely expect it to swap behaviors.  I have found these drones to be a lot of fun but not very precise.  The bebop line is far more precise than the mambo.