amymcgovern/pyparrot #142: multiple parrot mambo drone control

Shatad-Purohit, 2019-02-13:
I am trying to connect two parrot mambo drones, with my code. Initially I see that my laptops connects to both the drones (I see both the drone-lights stop blinking), but only one drone takes off. I am trying to control both the drones simultaneously (swarm scenario), can you help?

from pyparrot.Minidrone import Mambo

# If you are using BLE: you will need to change this to the address of YOUR mambo
# if you are using Wifi, this can be ignored
mamboAddr1 = "e0:14:2e:0b:3d:c8"
mamboAddr2 = "D0:3A:EE:55:E6:22"

# make my mambo object
# remember to set True/False for the wifi depending on if you are using the wifi or the BLE to connect
mambo1 = Mambo(mamboAddr1, use_wifi=False)
mambo2 = Mambo(mamboAddr2, use_wifi=False)

print("trying to connect")
success1 = mambo1.connect(num_retries=3)
success2 = mambo2.connect(num_retries=3)
print("connected: %s" % success1)
print("connected: %s" % success2)

if (success1):
    mambo1.ask_for_state_update()
    print("taking off1!")
    mambo1.safe_takeoff(5)

if (success2):
    mambo2.ask_for_state_update()
    print("taking off2!")
    mambo2.safe_takeoff(5)

if (success1):
    mambo1.safe_land(5)
    print("landing1!")
    mambo1.disconnect()

if (success2):
    mambo2.safe_land(5)
    print("landing2!")
    mambo2.disconnect()

amymcgovern, 2019-02-13:
You can only connect to one mambo at a time using BLE.  It says that in the documentation too.  It's an issue with the underlying BLE package.  You can control two using the same laptop but you have to use two separate scripts.

josefias, 2019-07-05:
hi there, when i run this code but change the 'safe_takeoff(5)' and 'safe_land(5)' to just 'takeoff()' and 'land()'. but 2 drones its de max that i have achieved.
but i came here to ask

, wy cant the BLE maintain more that that, shouldn't it be possible tto go up to 8 paired devices in BLE? 
Is it just because it wasn't designed for it?

amymcgovern, 2019-07-05:
The underlying BLE library I am using is limited to 1 per session.  Then BLE itself takes bandwidth.  It isn't infinite.  My pi would only sustain 8.

josefias, 2019-07-06:
where you able to fly 8 mambos at the same time?

amymcgovern, 2019-07-06:
Yes, I had them doing a simple program of taking off, flipping, and landing.  But I got 8 in the air at once, maximum.  6 was more reliable.

sergiyz57918, 2019-07-27:
Am I understanding correctly that BLE works only in Linux?

amymcgovern, 2019-07-27:
Correct.  The underlying BLE library that I use is only supported on linux.  If you can find a supported BLE library that is multi-os, I would happily upgrade.  At the time I wrote the library, such a multi-os BLE library did not exist.

sergiyz57918, 2019-07-28:
Did anyone try to use multiple Wi-Fi adapters in order to control multiple drones?

sergiyz57918, 2019-07-28:
https://wiki.qt.io/Main
https://www.qt.io/qt-for-python
Qt seems to be multiplatform BLE library

amymcgovern, 2019-07-28:
Qt is a GUI library. Not a BLE library...

ukBaz, 2019-07-28:
I also thought Qt was just a GUI library but @sergiyz57918 comments prompted me to investigate. 
I found a company update from July 1st 2015 that said:
https://www.qt.io/company-updates/qt-5-5-introduces-windows-10-support-adds-full-bluetooth-low-energy-official-support-for-red-hat-enterprise-linux-6-6

> This latest release also includes features such as Bluetooth Low Energy (Bluetooth Smart) especially beneficial for IoT (Internet of Things) deployments where it is now easy to write cross-platform code for communicating with smart sensors, wearables and other BLE-enabled peripherals. In addition, a new version of the integrated development environment, Qt Creator 3.4, enables more code and design productivity than ever before.

And some C++ examples for Bluetooth at:
https://doc.qt.io/qt-5/bluetooth-examples.html

Haven't found anything for Open Source or Python as of yet.

Do you have any links @sergiyz57918 ?

amymcgovern, 2019-07-28:
Well, we need something python :) That's why I chose the library I chose.  And the underlying protocol for BLE differs by OS. I looked into it deeply when I was deciding what to use.

ellisc1, 2019-09-24:
> I actually DID manage to connect multiple Mambo drones at the same time and have them fly in the air. The way I did this is I had another python file in the same project but in another directory that contained some code make my secondary Mambo take off. In my main code, I had my main Mambo launch sequence in a function by which I made a Thread. I then made another thread to call that python script that activated the second drone. The way I called that script was with pythons os library. I used os.command("python3 /mambo_1.py") and then it worked for both drones. I have the code in my repo, Swarm. Just thought that was so awesome I had to share.

Do you mind helping me out on what you did to make this work?