amymcgovern/pyparrot #143: Connecting multiple Bebop using a static IP?

KhristopherK, 2019-02-14:
Hi,

Following a different guide I have given a bebop a static IP within out network. I would like to know how to address that IP specifically within pyparrot. The idea is that I have multiple bebops and would like to independently address each one by it's IP Address.

Thanks,

amymcgovern, 2019-02-14:
Oh, how did you do that?  It should be easy to add in a hook for a non-default IP (I just have it hard-coded right now).  But I would love to do this.  Found some directions on the parrot dev board but I hadn't tried them yet (wasn't a high priority yet).

KhristopherK, 2019-02-15:
We followed this link -> https://github.com/tnaegeli/multiple_bebops
Could you tell us how pyparrot get the IP address? our plan today is to get them working :)

amymcgovern, 2019-02-15:
pyparrot uses mDNS to get the IP address (except for the latest mambos because parrot broke their mDNS server on the mambos in one of the latest firmware updates).  The code is in here:

https://github.com/amymcgovern/pyparrot/blob/master/pyparrot/networking/wifiConnection.py

When you changed your IP address, I suspect it didn't update the mDNS inside pyParrot.  One could edit that code to take an IP address in as an optional parameter instead of relying on mDNS.

amymcgovern, 2019-02-19:
@KhristopherK I just put in a code update that should allow this.  I haven't tested other than to ensure the default still works as I haven't hacked my bebop yet :)

KhristopherK, 2019-02-20:
Hi @amymcgovern thanks!, I can confirm it works on non-standard IP addresses. What I will try next is connecting 3 and trying to get video from them. I will keep you posted.