amymcgovern/pyparrot #144: how can I take photo with bebop2?

EagleX1995, 2019-02-19:
bebopVision.open_video() won't return current images. All I get were the old images.

amymcgovern, 2019-02-19:
Did you read the documentation and the sample code?  That is not how you get images.

https://pyparrot.readthedocs.io/en/latest/vision.html

EagleX1995, 2019-02-19:
I had run the sample code demoBebopVision.py
The program returned the images from the previous execution.

amymcgovern, 2019-02-19:
What OS are you on?  sounds like your ffmpeg program didn't clean out the old images.

EagleX1995, 2019-02-19:
The latest macOS. So please tell me how can I clean out the old images. Thank you very much.

amymcgovern, 2019-02-19:
that makes it easy!  It should be deleting them automatically though.  Since Mac OS is my primary environment, I'll go look into that.  In the meantime, find the images directory which should be inside your pyparrot installation and remove everything in there.

EagleX1995, 2019-02-19:
Thank you. This is really helpful. Feels like removing all images under the directory after each execution can be the simplest solution for me. I was keeping rebooting and reseting my bebop LOL

amymcgovern, 2019-02-19:
Yeah this is definitely in the directory and not on the bebop since you are using ffmpeg.  If you use the demoBebopVisionGUI version that won't happen (which is what I use 99% of the time).

amymcgovern, 2019-03-02:
I fixed this today.  ffmpeg now cleans out the images directory each restart by default (but it can be turned off if you prefer that).

amymcgovern, 2019-03-02:
Also fixed the bug where ffmpeg wasn't being killed properly!