amymcgovern/pyparrot #145: Mambo altitude sensor errors

walber000, 2019-02-28:
Hello Amy,

I'm doing some testing with my Mambo's altitude readings.
Initially it seems to be a good sensor, but I observed three cases when it shows wrong measurements:
- If you move something below the drone (when it is just hovering and keeping position), it tries to "correct" its altitude going up, but this extra height is not showed in mambo.sensors.altitude, so the drone is higher than the measurement says.
- Sometimes when moving downward slowly (from 0.8m to 0.5m for example) and verifying altitude, the measurements starts to get wrong, and the drone goes lower than expected.
- If I apply an horizontal movement (pitch for example) for a few seconds, the drone sometimes moves a little vertically too, but again this height difference does not modify the readings, so the drone is actually higher or lower than the sensor shows.
So it seems to me that the ultrasonic sensor is not actually measuring height by sending and receiving sonic pulses. Do you know something about this? May you please test this in your own mambos and check this problem?

amymcgovern, 2019-03-01:
For the first one:  I have observed this in our mambos.  This is happening inside the firmware and not in our software.  

For the second one:  the sensor is only sending data at 2Hz so it can get pretty far off if you are moving quickly.

For the third one: doesn't surprise me.

I've found the altitude to be a pretty bad sensor honestly.  I try to avoid using it!

walber000, 2019-03-01:
Thank you very much for the explanation. I will use an external camera to measure the drone altitude.