amymcgovern/pyparrot #148: Mambo with sched schedular not working

ihsancemil, 2019-03-05:
```
sched.scheduler(time.time, self.mambo.smart_sleep).enter(10, 2, self.mambo.fly_direct, (20, 0, 0, 0, 1))
sched.scheduler(time.time, self.mambo.smart_sleep).enter(40, 2, self.mambo.fly_direct, (20, 0, 0, 0, 1))
sched.scheduler(time.time, self.mambo.smart_sleep).enter(70, 2, self.mambo.fly_direct, (-20, 0, 0, 0, 1))
sched.scheduler(time.time, self.mambo.smart_sleep).enter(100, 2, self.mambo.fly_direct, (-20, 0, 0, 0, 1))
```

I want to send roll, pitch, yaw angles to the parrot mambo for every 30 second to see its mathematical behavior. However If I write a code like above pyparrot giving an error with description  

>  timeout - trying again

This error sometimes solving by itself without writing any correction code, however generaly it gives me an error.

amymcgovern, 2019-03-05:
If you are using time.sleep in there at all, it won't work.  The BLE disconnects with any use of time.sleep.  it's why I call smart_sleep.  Now the wifi shouldn't care...

ihsancemil, 2019-03-06:
I'm using wifi connection. I'm trying to track an object with image processing and control theory. Is there any way to print out timeout in wifi mode?

ihsancemil, 2019-03-06:
Also I'm using Parrot Mambo Version 1.5.7, is there any change to get over this problem by updating the library?

amymcgovern, 2019-03-06:
Updating the library is always a good thing IMHO.  I don't usually see timeouts in wifi mode (but I can guarantee them in BLE due to the way the BLE library works).  I do have an automated timer that tries to reconnect.  Do the mambo eyes switch to blinking when this happens?  Maybe you have an error elsewhere in the code causing the disconnect?

ihsancemil, 2019-03-08:
Yes mambo starts blinking when I try to send data and I get timeout. Also what did you mean bu "automated timer that tries to reconnect". Should I call mambo.connect over and over again to escape timeout error?

ihsancemil, 2019-03-08:
It seems that even 
`self.s = sched.scheduler(time.time, self.mambo.smart_sleep)`
breaks wifi connection. Writing my own scheduler algorithm solved the problem.

I simply initialize a list with tasks inside (second, function, args, is_done)
```
                self.task_list = [[10, self.mambo.fly_direct, (20, 0, 0, 0, 1), False],
                                  [40, self.mambo.fly_direct, (20, 0, 0, 0, 1), False],
                                  [70, self.mambo.fly_direct, (-20, 0, 0, 0, 1), False],
                                  [100, self.mambo.fly_direct, (-20, 0, 0, 0, 1), False]]
```
then I perform if second claimed

```
            second = time.time() - self.start_time
            print("Time get: ", second)
            for task_index in range(len(self.task_list)):
                if not (self.task_list[task_index][3]) and (second -1 < self.task_list[task_index][0] < second + 1):
                    print("Performing task: ", task_index)
                    self.task_list[task_index][3] = True
                    self.task_list[task_index][1](*self.task_list[task_index][2]) 
```
even if it is spagetti code, this is enough for me for now.