amymcgovern/pyparrot #151: findMinidrone: command not found

SvengalH, 2019-03-08:
Development environment: Ubunto 14.04 LTS (this is old. Could that be the issue)
I have a feeling the issue is that I don't really understand the directory structure and path requirements surrounding python. Any suggestions for an outside resource to consult on this would be welcomed.

SvengalH, 2019-03-08:
Sorry. I somehow posted prior to completing the write up of my issue. Here is some supporting info.

Problem:
`me@prompt:~/anaconda3/bin$ sudo findMinidrone
sudo: findMinidrone: command not found
`
Related packages installed:
`me@prompt:~/anaconda3/bin$ dpkg -l | grep -iE 'python-pip|bluez|bluetooth|libglib2.0-dev|python-bluez'
ii  bluetooth                                             4.101-0ubuntu13.3                                    all          Bluetooth support
ii  bluez                                                 4.101-0ubuntu13.3                                    amd64        Bluetooth tools and daemons
ii  bluez-alsa:amd64                                      4.101-0ubuntu13.3                                    amd64        Bluetooth ALSA support
ii  bluez-cups                                            4.101-0ubuntu13.3                                    amd64        Bluetooth printer driver for CUPS
ii  bluez-gstreamer                                       4.101-0ubuntu13.3                                    amd64        Bluetooth GStreamer support
ii  gir1.2-gnomebluetooth-1.0                             3.8.2.1-0ubuntu4.2                                   amd64        Introspection data for GnomeBluetooth
ii  gnome-bluetooth                                       3.8.2.1-0ubuntu4.2                                   amd64        GNOME Bluetooth tools
ii  indicator-bluetooth                                   0.0.6+14.04.20140207-0ubuntu2                        amd64        System bluetooth indicator.
ii  libbluetooth3:amd64                                   4.101-0ubuntu13.3                                    amd64        Library to use the BlueZ Linux Bluetooth stack
ii  libglib2.0-dev                                        2.40.2-0ubuntu1.1                                    amd64        Development files for the GLib library
ii  libgnome-bluetooth11                                  3.8.2.1-0ubuntu4.2                                   amd64        GNOME Bluetooth tools - support library
ii  pulseaudio-module-bluetooth                           1:4.0-0ubuntu11.1                                    amd64        Bluetooth module for PulseAudio sound server
ii  python-bluez                                          0.18-2                                               amd64        Python wrappers around BlueZ for rapid bluetooth development
ii  python-pip                                            1.5.4-1ubuntu4                                       all          alternative Python package installer
ii  python-pip-whl                                        1.5.4-1ubuntu4                                       all          alternative Python package installer`

Related Python Packages installed:
`me@prompt:~/anaconda3/bin$ pip3 list | grep -iE 'pyparrot|untangle|zeroconf|bluepy'
bluepy (1.3.0)
pyparrot (1.5.15)
untangle (1.1.1)
zeroconf (0.21.3)
`
python3 is installed:
`me@prompt:~/anaconda3/bin$ which python3
/home/max/anaconda3/bin/python3
`

amymcgovern, 2019-03-11:
That is odd.  It should appear in a bin directory.  You can find the code directly here:

https://github.com/amymcgovern/pyparrot/tree/master/pyparrot/scripts

SvengalH, 2019-03-11:
Thanks. I'll take another poke at this as time allows this week.

SvengalH, 2020-01-25:
Life interrupted my progress on this, but now I'm back at it. Since I last posted, I got a modern laptop, learned a bit about setting up Python with virtual environments and reinstalled pyparrot. I'm still a bit green though. In my new installation I have the same problem I had previously.

**Problem:**
I suspect I'm not doing an installation step that is so fundamental, that it is assumed rather than explicitly spelled out in the instructions. Maybe something to do with Github. I simply downloaded the zip file and extracted it to my virtualenv directory. Am I supposed to use a fancy git command instead?

The findMinidrone script is missing from the bin directory, so I wonder what I am not doing right.

```
(pyparrot) me@laptop:~/.virtualenvs/pyparrot$ find -name findMinidrone.py
./lib/python3.7/site-packages/pyparrot/scripts/findMinidrone.py
./pyparrot/scripts/findMinidrone.py
```

I can change to the scripts directory and succesfully run findMinidrone.py 
```
(pyparrot) me@laptop:~/.virtualenvs/pyparrot/pyparrot/scripts$ sudo python3 findMinidrone.py 
Discovered device e0:14:a9:8b:3d:fc
...
Discovered device 50:a2:5d:8c:73:c1
FOUND A MAMBO!
Device e0:14:69:85:3d:fc (random), RSSI=-30 dB
  Complete Local Name = Mambo_528002
```

**Development environment:**
Ubuntu 19.04, Python 3.7.3, MS Studio Code, virtualenv package and virtualenvwrapper python package.

**Related packages installed (same as before):**
```
ii   bluetooth                   5.50-0ubuntu2.1      all             Bluetooth support
ii   bluez                       5.50-0ubuntu2.1      amd64      Bluetooth tools and daemons
ii  bluez-cups                   5.50-0ubuntu2.1      amd64      Bluetooth printer driver for CUPS
ii  bluez-obexd                  5.50-0ubuntu2.1      amd64      bluez obex daemon
ii  gir1.2-gnomebluetooth-1.0:amd64  3.32.1-1         amd64      Introspection data for GnomeBluetooth
ii  gnome-bluetooth              3.32.1-1             amd64      GNOME Bluetooth tools
ii  libbluetooth3:amd64          5.50-0ubuntu2.1      amd64      Library to use the BlueZ Linux Bluetooth stack
ii  libglib2.0-dev:amd64         2.60.4-0ubuntu0.19.04.1  amd64  Development files for the GLib library
ii  libglib2.0-dev-bin           2.60.4-0ubuntu0.19.04.1  amd64  Development utilities for the GLib library
ii  libgnome-bluetooth13:amd64   3.32.1-1             md64       GNOME Bluetooth tools - support library
ii  pulseaudio-module-bluetooth  1:12.2-2ubuntu3.1    amd64      Bluetooth module for PulseAudio sound serve
ii  python-bluez                 0.22+really0.22-1    amd64      Python 2 wrappers around BlueZ for rapid bluetooth development
ii  python-pip                   18.1-5               all        Python package installer
ii  python-pip-whl               18.1-5               all        Python package installer
```
**Related Python Packages installed (same as before):**
```
(pyparrot) me@laptop:~/.virtualenvs/pyparrot/$ pip3 list | grep -iE 'pyparrot|untangle|zeroconf|bluepy'
bluepy        1.3.0   
pyparrot      1.5.20  
untangle      1.1.1   
zeroconf      0.24.4 
```
For what its worth, I am able to change to the examples directory and successfully run demoMamboDirectFlight.py