amymcgovern/pyparrot #155: no module named pyparrot.Minidrone

natopotato34, 2019-03-19:
I keep getting "no module named pyparrot.Minidrone" when I run any of the demo files that include the "from pyparrot.Minidrone import Mambo" (see photo). If I go into the scrpits of pyparrot however, I can see Minidrone there in the directory. Any advice?

[image]

amymcgovern, 2019-03-19:
Install it using pip or use the setup.py.  That sets up your python paths. There are instructions in the documentation.

InquisitiveAS, 2019-11-04:
What does this mean ! I got the same issue

$ sudo python setup.py
/usr/lib/python3.6/distutils/dist.py:261: UserWarning: Unknown distribution option: 'long_description_content_type'
  warnings.warn(msg)
usage: setup.py [global_opts] cmd1 [cmd1_opts] [cmd2 [cmd2_opts] ...]
   or: setup.py --help [cmd1 cmd2 ...]
   or: setup.py --help-commands
   or: setup.py cmd --help

error: no commands supplied

amymcgovern, 2019-11-04:
I assume you need to do install (e.g. setup.py install) but I honestly never use that route. I do pip install.