amymcgovern/pyparrot #156: Gps coordinate to start and end to make a fligth plan

kevin99j, 2019-03-21:
Good day, and thanks for your work!!
how I can make a fligth plan giving coordinates to start and end the fligth I am new in python and i cant understand how parse the sensor info or how that retrieve this info
thanks for your reply, Kevin.

amymcgovern, 2019-03-21:
the sensor info for the bebop is all in a dictionary.  commonly used ones also have their own variable but everything is in a dictionary.  So just print out the dictionary to find the key/values that you want and proceed with using those values.  Most are updated at 10 Hz, which is the frequency that the firmware broadcasts them.  bebop.sensors.sensors_dict is the name of the dictionary.

kevin99j, 2019-04-23:
the sensor dictionary return 
{ 'GpsLocationChanged_altitude_accuracy': 0, 'GpsLocationChanged_latitude': 500.0, 'PositionChanged_altitude': 500.0, 'GpsLocationChanged_longitude_accuracy': 0, 'AttitudeChanged_roll': 0.004713607951998711, 'PositionChanged_longitude': 500.0, 'PositionChanged_latitude': 500.0}
how i can unlock gps to get this information, in the app free fligth pro i can get gps coordinates, and i want to get this info, can you help me please?

amymcgovern, 2019-04-23:
Are you outside or inside?  the GPS just needs to get a lock.  It always starts at those coordinates but it will update when it gets a lock (which is automatic).

kevin99j, 2019-05-18:
How can I update it? , the flight is outside but I always receive the same coordinates, can you show me a code fragment where you get the GPS information with real coordinates ?, I appreciate your help

amymcgovern, 2019-05-23:
How are you accessing it?  It changes every 0.1 of a second for me.