amymcgovern/pyparrot #157: MAVLink File vs. Send Controller GPS in xml files

CZRam, 2019-03-23:
Hi,

First, huge thanks for making this library available. I found among the documentation under `/commandsandsensors` a [common.xml] (https://github.com/amymcgovern/pyparrot/blob/master/pyparrot/commandsandsensors/common.xml#L621) document that contains commands to initiate a mavlink flight plan from a file. How is this accessible from the current library? Alternatively, I also noticed what looked like an ability to send GPS coordinate commands to the drone in [ardrone3.xml](https://github.com/amymcgovern/pyparrot/blob/master/pyparrot/commandsandsensors/ardrone3.xml#L2596) - is that possible at all? How are these XML files used?

Thanks again,
Carlos

amymcgovern, 2019-03-23:
You could add commands to do that, yes.  The xml files are parsed and used to send commands.  It would be pretty easy to add such a command.

amymcgovern, 2019-03-23:
the xml files themselves come from parrot's SDK

CZRam, 2019-03-23:
Is adding support for this something on the roadmap? Or could you point me to some documentation that would help me implement those functions in a PR?

amymcgovern, 2019-03-23:
Just look at any of the commands inside bebop.  They all use the parser.  It is intended to make it easy to send the commands.  for example, fly_direct takes arguments.  flip handles enums.  Etc.