amymcgovern/pyparrot #159: Streaming Fisheye video / pictures

MoiAlex, 2019-03-27:
Hi Amy, thanks a lot for such a amazing work !
My ultimate goal is to fly the bebop2 and stream video with the FishEye format to my computer with Python and do some real-time image processing.

I use your demoBebopVision.py and it is working great on my python3.5.
But even if I change the pictures format to jpeg_fisheye with the line : 
` bebop.set_picture_format('jpeg_fisheye') `
I still have pictures in a jpeg classic format. Reading the Bebop doc, I am guessing that it is impossible to record a video with the Fisheye format (only pictures). 

In this issue https://github.com/amymcgovern/pyparrot/issues/49#issuecomment-411101707, you suggest that it is possible to save the current image.

> You just want to save a photo from the camera? Easy. Just ask it for the current image at the time and then save it.

The problem is that the stream opened with Ffmpeg is a video and so it is impossible to use the Fisheye format. So I want to avoid using video but picture instead.

I implemented the `take_picture` function for Bebop, which is working fine with Fisheye format when I check the files in the Bebop with my phone. Now I need to find a way to receive them on Python. I tried to write the Bebop functions I need with the help of `class MamboGroundcam` but I couldnot make them work. 

And so simple first question, how can I receive a single picture and not the full stream opened with Ffmpeg ? How can I transfer files saved on the Bebop to my computer ? 
 
Second question, do you think it would it be possible to stream in Fisheye format by taking enough pictures (hopefully at least 20fps) and not a video and send them to the computer ? 

I hope I am clear.
Thanks in advance for your help !
Alex

amymcgovern, 2019-03-27:
That is a very interesting idea.  I never tried using the fisheye format.  I looked through the commands in ardrone3 and don't see a download command for the photos.  Based on personal experience of flying our own bebop and using photos (outside of the autonomous stuff we do), I doubt you can get a 20 fps rate.  As for how to download it, I don't actually know.  Have you searched on the parrot development forum?

MoiAlex, 2019-03-29:
I am quite sure know that it is not possible to take a video with the Fisheye format. This format is only supported for pictures as the Parrot Developers suggest. Moreover you are right, it looks like it needs a long time to take and save pictures on the bebop so I do not think either that it would be possible to take pictures with 20fps rate... I am going to have a try anyway, I'll let you know.

I have implemented a "real-time" (delay ~300/400ms) image processing on the stream of the bebop. For that, I had a lot of issues when I was processing the image in the UserVision callback function like you suggested. So I process now the image ouput of the line `img = cv2.imread(self.file)`, in the ` _buffer_vision ` function, from the class `DroneVisionGUI`. It is not proper I guess but still helpfull to start. I will have a look to improve this too.

Ref to the Parrot SDK doc, it seems that the `libARDataTransfer` allows us to transfer data saved in the Bebop. Have you ever try to anything like this ? 

> Download pictures and videos
> Once the picture or video has been taken, the Bebop stores it on its internal memory. Pictures are stored in `internal_000/Bebop_Drone/media/.`
> To get the pictures, you can:
> - simply do a ftp request:
>         - port: 21
>         - login: “anonymous”
>         - no password
> - use libARDataTransfer which provides an abstraction of the ftp
> 
> Here is how to do it with libARDataTransfer.
> libARDataTransfer lets you get a list of all stored medias quite quickly. It also provides you a way to enrich the list of medias with their thumbnails. It also gives you the ability to download the media.
> 

Thanks for your work
Alex

amymcgovern, 2019-03-29:
If it is ftp, it works like the mambo and we could adapt the mambo interface.  They broke the mambo interface on their last firmware upgrade though.  But the code to do essentially that is in the mambo and would be really easy to adapt.

itssme, 2019-07-03:
>  I am quite sure know that it is not possible to take a video with the Fisheye format.

[image]

Actually it is quite possible to change the video output of the drone to a different format. Thanks to [this hacking guide](http://fargesportfolio.com/wp-content/uploads/2018/01/BeebopHackingGuide1_7_2.pdf) we know it is. As you can see in section "4.4.5 Custom camera settings" it is possible to change the FOV of the camera as well as the resolution.

Utilising the "dragon-prog" program which basically controls your drone you can adjust some settings including the FOV. However unfortunately since firmware "3.3.0" those options no longer exist because parrot decided to remove them :/ So you will have to downgrade to "3.2.0" to re-enable those settings.

## How to downgrade:
Before you change anything I have to warn you that I did not try this myself _**yet**_ and I can't guarantee that this will work so please keep that in mind if you choose to downgrade. It is also worth taking a look in at section "5.4" in the bebop hacking guide.

1. Download 3.2.0 Firmware for Bebop2 from [this link](https://drivers.softpedia.com/get/drone-helicopter-multicopter-controller/Parrot/Parrot-Bebop-2-Drone-Firmware-320.shtml)
   - If you have a Bebop1 use [this link](http://uploaded.net/file/8jpu4mmp/from/h4fwze)

### Use a USB Stick to downgrade (only bebop2)
- I am not sure if this will work but its probably worth to give it a try

1. Format a USB Drive with FAT32 and put the file on the stick
     - make sure this is the only file on the stick
1. Connect the USB stick to your bebop (you will require a microusb to usb cable)
1. It should automatically downgrade to "3.2.0" and could take up to 15 min or so

### Manually downgrading
1. Rename the file to bebop2_update.plf (or bebop1 if you use one)
1. Plug the bebop into your PC via a Micro-USB cable
1. Use File-Zilla to connect to 192.168.1.42
1. Drop the file to `internal_000` and close File-Zilla
1. Connect to the drone either via telnet or adb-shell. (under windows use something like putty)
`telnet 192.168.42.1` or
`adb connect 192.168.43.1:9050`
1. Then run the command `mount -o remount ,rw /` which will make the filesystem writable
1. Open the `dgwl.txt` file in vi: `vi /bin/updater/dgwl.txt`
1. add the version 3.2.0 to the file
1. Then start the update `/bin/updater/updater_scan.sh /data/ftp/internal_000`

## Set the video resolution an FOV
- If the downgrade works connect to the drone using telnet or adb shell und run this command `/usr/bin/dragon-prog -c 1280x3072 -s 1024x600 -V 140`. It will set the FOV to 140°

NOTE: The newest version of free-flight probably won't work anymore and you will have to search for an older one.

closed 2019-07-03 by amymcgovern