amymcgovern/pyparrot #161: Mambo: Failed to connect with rtsp://192.168.99.1:554/media/stream2

BlakeList, 2019-03-31:
Hi there,

I am currently trying to get the video stream working for the Parrot Mamob using VLC, but I am running into the following error when trying to run the demoMamboVisionGUI.py code. I am using Windows10 64bit and running everything with sublime and anaconda. I have VLC 3.0.6 and all dependencies installed.

`[000001fbc55fb950] live555 demux error: Failed to connect with rtsp://192.168.99.1:554/media/stream2

[000001fbc560e170] main stream error: connection failed: Connection refused by peer

[000001fbc560e170] satip stream error: Failed to connect to RTSP server 192.168.99.1:554

[000001fbc560e170] main stream error: connection failed: Connection refused by peer

[000001fbc560e170] access_realrtsp stream error: cannot connect to 192.168.99.1:554

[000001fbc560e170] main stream error: Connection failed

[000001fbc560e170] main stream error: VLC could not connect to "192.168.99.1:554".

[000001fbc5639a20] main input error: Your input can't be opened

[000001fbc5639a20] main input error: VLC is unable to open the MRL 'rtsp://192.168.99.1/media/stream2'. Check the log for details.
`

I have all the correct packages installed and can get the drone working with the other demo code to get it to move/flip. I am wanting to stream the video to process for image detection.

I should also add that I am on the university WiFi but I don't expect this to be a problem if I am connecting locally to the drones IP. 

Please help!

Thank you for your time.

amymcgovern, 2019-03-31:
Your last line is the issue.  You cannot be on the university wifi.  You have to be on the drone's wifi network to make it work.

BlakeList, 2019-03-31:
Hi Amy,

Thank you for replying so fast. My apologies, I should clarify, I am connected to the Mambo WiFi, hence why I can fly the Mambo with the code. I am just unable to get the video stream out.

amymcgovern, 2019-03-31:
Glad to hear it!  I assume streaming is working fine to your phone?  What version of the firmware?  That is very odd.

BlakeList, 2019-03-31:
Yes, just tested again. The video stream works through my phone. Can take pictures and record video. Drone software version is 3.0.26.

amymcgovern, 2019-04-01:
My best guess at this point (with current firmware and current pyparrot) is to reboot the camera but I'm assuming you have done that.  Every once in awhile at a workshop, we have had bad cameras that literally are broken out of the box.  Out of curiosity, is your wifi network showing up as a Mambo_somenumbers?  The broken cameras all seem to NOT do that.

BlakeList, 2019-04-16:
Sorry for the late reply. I now have two Mambo drones with working cameras (can view video through phone app) but still no luck getting ffmpeg or vlc to work. Like I said, I can connect to the drone through the computer via WiFi and can implement the demo flight code (take off, flip, land, etc.) but cannot get the video out. 
Would you happen to know of an implementation for getting the video for Windows?

Thanks for your time.

amymcgovern, 2019-04-16:
That is very odd!! Same error as above?  I have tested it on windows plenty of times.  Have you tried the ffmpeg version instead of VLC?

BlakeList, 2019-05-09:
Hi Amy,

I just fixed the issue. The problem was that the ip address: 192.168.99.1 was being used by the adapter in Virtual Box. This was found using `ipconfig` in the command prompt. I simply had to change the ip address in VB to something else so that it could be used for the connection with the drone.

You may close this issue now. Thank you very much for your help. I will post again with my code at the end of my project working on object tracking/following using Tensorflow.

Thanks!
Blake

amymcgovern, 2019-05-09:
That would do it!

closed 2019-05-09 by amymcgovern