amymcgovern/pyparrot #162: Quick Start Demo Code for Disco

Disco1234, 2019-04-02:
I saw that Disco was added to WiFi connected devices a few weeks ago and was wondering if there was any Quick Start Demo Code similar to the Quick Start Demo Code for Mambo and Bebop.  I've been tinkering with those scripts while I have my Raspberry Pi WiFi connected to my Disco, but so far I haven't had any success.  If I could get some parameters like altitude, latitude, longitude, then assign them to variables and print them, I'd know I'm pulling real data from the Disco aircraft.

Does anyone know if Disco has BLE?  If so, how do you determine its address?

amymcgovern, 2019-04-02:
I don't actually have a disco so the support for it was added by someone else.  Thus I don't have quick connect code I can demo.  I just took the PR as it looked like the code was a quick fix (it was just looking for a different mDNS address, which is how parrot runs their devices).  The rest should be similar.

Disco1234, 2019-04-02:
Thanks for the response Amy, I'll keep experimenting

Disco1234, 2019-04-03:
As a workaround, I replaced the Bebop2 mDNS address in wifiConnection.py with the Disco address.  I was then able to connect using the Bebop demo code.  Then I noticed that I couldn't connect the SkyController or FreeFlight Pro app to the Disco while my Raspberry Pi was connected (and vice versa).  Is there a way to have them all connected simultaneously?

amymcgovern, 2019-04-04:
No, their firmware only allows one connection.

Disco1234, 2019-04-04:
Ah, thanks.  Do you know if it's possible to modify the firmware to allow more than one connection?

amymcgovern, 2019-04-04:
I haven't looked into the firmware at all so I don't know.  I know on the protocol end, it does a handshake.  If someone else is already connected, the handshake fails.

Disco1234, 2019-04-05:
Got it, thanks.  I ask because I was thinking of having a companion computer connected while the Disco executes a Flight Plan.  I might try and look into whether the companion computer could communicate with the CHUCK through its USB ports.