amymcgovern/pyparrot #166: Mambo connection/movement problem (bluetooth)

walber000, 2019-06-14:
Hello Amy,

Is there a way to receive an ack signal from the drone when sending piloting commands with fly_direct?

I have a previous issue that the drone's pitch movement tends to one side, but I verified that different tests give me different results, with the same code.
I understand that's normal, but sometimes a command with pitch=10 and roll=30 (max_tilt = 10) takes a long time to make the drone move to the right,
and sometimes it responds quickly, so I'm thinking maybe it's a connection issue, because the drone moves away from the bluetooth dongle (about 3 meters),
and the movement itself may disrupt the connection.
(Note: this problem (and the "bias to the left" problem) do not happen when piloting the drone manually with a smartphone)

If I could check what the drone is receiving, maybe I could solve that.

amymcgovern, 2019-06-14:
Not that I am aware of, alas!

walber000, 2019-06-14:
If I send a flying command, and after that I request a sensor reading, if I receive the reading, maybe I can affirm that the flying command was understood and done?

amymcgovern, 2019-06-17:
Maybe?  There just isn't an ack that I know of.  It uses UDP and not TCP in the underlying protocol.