amymcgovern/pyparrot #169: Bebop ffmpeg vision demo  not working

Yeb02, 2019-08-13:
I am trying to get pictures live from the bebop2 and followed the docs. It went well until the "Bebop ffmpeg vision demo" code, which crashes. I get the following error:
bebop.disconnect()
Traceback (most recent call last):
  File "<console>", line 1, in <module>
NameError: name 'bebop' is not defined

>>> (executing file "pyparrot2.py")
Setting up mDNS listener since this is not a Mambo
Making a browser for _arsdk-090c._udp.local.
Service Bebop2-038622._arsdk-090c._udp.local. added, service info: ServiceInfo(type='_arsdk-090c._udp.local.', name='Bebop2-038622._arsdk-090c._udp.local.', addresses=[b'\xc0\xa8*\x01'], port=44444, weight=0, priority=0, server='Bebop2-038622.local.', properties={b'{"device_id":"PI040384AG6I038622"}': False})
{"d2c_port": 43210, "controller_type": "computer", "controller_name": "pyparrot", "arstream2_client_stream_port": 55004, "arstream2_client_control_port": 55005}
{'status': 0, 'c2d_port': 54321, 'c2d_update_port': 51, 'c2d_user_port': 21, 'qos_mode': 0, 'arstream2_server_stream_port': 5004, 'arstream2_server_control_port': 5005}
c2d_port is 54321
starting listening at 
 Success in setting up the wifi network to the drone! 
51
c:\programdata\anaconda3\lib\site-packages\pyparrot\images
c:\programdata\anaconda3\lib\site-packages\pyparrot\utils
removing all the old images
Opening ffmpeg
ffmpeg -protocol_whitelist "file,rtp,udp" -i c:\programdata\anaconda3\lib\site-packages\pyparrot\utils/bebop.sdp -r 30 image_%03d.png
starting vision thread
Opening non-blocking readers
'ffmpeg' isn't recognized as an internal command
Traceback (most recent call last):
  File "C:\Users\alpha\Desktop\Informatique\TIPE\pyparrot\pyparrot2.py", line 36, in <module>
    success = bebopVision.open_video()
  File "c:\programdata\anaconda3\lib\site-packages\pyparrot\DroneVision.py", line 146, in open_video
    line_str = line.decode("utf-8")
UnicodeDecodeError: 'utf-8' codec can't decode byte 0x82 in position 27: invalid start byte
timeout - trying again
timeout - trying again
timeout - trying again

and it keeps looping on "timeout - trying again"

What did I do wrong ?
Thanks,

amymcgovern, 2019-08-13:
No, there is not a photo command for the bebop.  Did you install ffmpeg?  That's the error I see.

Yeb02, 2019-08-14:
I did it twice to be sure, once with pip and then with the official website. But nothing changes. Is this error related to the file's path (directory names) ? 
https://stackoverflow.com/questions/22216076/unicodedecodeerror-utf8-codec-cant-decode-byte-0xa5-in-position-0-invalid-s
Shall I edit the pyparrot code this way ?

amymcgovern, 2019-08-14:
The error is here:

'ffmpeg' isn't recognized as an internal command

amymcgovern, 2019-08-14:
NOT the utf error.  ffmpeg never starts!  You need it to be in your python path somehow.

Yeb02, 2019-08-17:
Thank you it finally worked ! The issue was with the ffmpeg release, that had to be zeranoe's.

Yeb02, 2019-08-18:
well I've had another problem: 
[h264 @ 00000283ab0dccc0] decode_slice_header error
[h264 @ 00000283ab0dccc0] non-existing PPS 0 referenced
is looping in the shell.

amymcgovern, 2019-08-18:
ffmpeg prints out a LOT of messages.  And sometimes it takes awhile to get a keyframe.  I'd say just be patient.  They are sent every 5-10 seconds and I've seen the stream take ~20 to start sometimes. Since you are on a bebop, the best thing is to move to 5Ghz if your laptop supports it.  It drops a LOT less frames than 2Ghz.

closed 2019-08-19 by Yeb02