amymcgovern/pyparrot #170: Bebop2 with libVLC

Yeb02, 2019-08-19:
Hi, I've had some issues with the libVLC demo code for the Bebop (using bebop 2 and windows 10)
Everything goes well excepted  the function  demo_user_code_after_vision_opened()  . It doesnt run at all: Ive added several "print" to check where it stops, and there is the result:

Setting up mDNS listener since this is not a Mambo
Making a browser for _arsdk-090c._udp.local.
Service Bebop2-038622._arsdk-090c._udp.local. added, service info: ServiceInfo(type='_arsdk-090c._udp.local.', name='Bebop2-038622._arsdk-090c._udp.local.', addresses=[b'\xc0\xa8*\x01'], port=44444, weight=0, priority=0, server='Bebop2-038622.local.', properties={b'{"device_id":"PI040384AG6I038622"}': False})
{"d2c_port": 43210, "controller_type": "computer", "controller_name": "pyparrot", "arstream2_client_stream_port": 55004, "arstream2_client_control_port": 55005}
{'status': 0, 'c2d_port': 54321, 'c2d_update_port': 51, 'c2d_user_port': 21, 'qos_mode': 0, 'arstream2_server_stream_port': 5004, 'arstream2_server_control_port': 5005}
c2d_port is 54321
starting listening at 
 Success in setting up the wifi network to the drone! 
connected to bebop
51
c:\programdata\anaconda3\lib\site-packages\pyparrot\images
c:\programdata\anaconda3\lib\site-packages\pyparrot\utils
51
<PyQt5.QtGui.QPixmap object at 0x0000019A44D7FD68>
QCoreApplication::exec: The event loop is already running
True
Starting user vision thread
Starting user drawing thread
success from play call is 0 
user callback being called
saving picture (720, 1280, 3)
saving picture (720, 1280, 3)
saving picture (720, 1280, 3)
saving picture (720, 1280, 3)
saving picture (720, 1280, 3)
saving picture (720, 1280, 3)
saving picture (720, 1280, 3)
saving picture (720, 1280, 3)
saving picture (720, 1280, 3)
saving picture (720, 1280, 3)
saving picture (720, 1280, 3)

and a single cv2 window opens, displaying a picture from the drone (with unhomogeneous definition, bugs, ....).  Despite saying picture saved (I have moved the print so that it is shown when a picture is saved)  there is nothing else in pyparrot\image than a visionStream.jpeg picture. What have I done wrong ?

amymcgovern, 2019-08-19:
Glad you fixed it...

Yeb02, 2019-08-20:
Do you have stream quality issues with your Bebop 2 ?

amymcgovern, 2019-08-20:
Not on 5Ghz.  On 2Ghz, for sure.  Move the Bebop to 5Ghz with the phone app.

Yeb02, 2019-08-28:
I did, but im still getting pictures like the following :

[image]

The size of the blurred stripe hasnt even changed. Should I reset my drone ?
(If it helps, the blurred part is actualized every 1 or 2 second during the stream, unlike the rest (25 fps). 
Thank you,

amymcgovern, 2019-08-29:
I definitely would try a reset.  Do you get the same strange area on your phone when you look at the video?  Have you recorded from the drone and is it showing there?  Almost like your camera isn't sending the last part right.  Definitely not what I see.  I see dropped frames a lot but not that!

amymcgovern, 2019-08-29:
p.s. if you reset you have to go back to 5Ghz again.

Yeb02, 2019-08-30:
The reset didnt change anything, and videos with freeflight pro are fine. I'll try with other versions of VLC.

Yeb02, 2019-09-14:
Even using the oldest working versions (like 3.0.2) didn't solve it. Could it have something to do with the bebop's software version ?

amymcgovern, 2019-09-14:
I really don't know :( I've never seen that issue on mine at all!

Yeb02, 2019-10-20:
I created a linux (ubuntu 18.04) partition on the same computer and the bug disappeared. Really don't understand the problem since the versions are the same, but well, it works.

nigelveach, 2020-05-06:
Hi! I am having exactly the same issue (a blurry bar at the bottom of the VLC).  Did either of you discover a fix to this problem, other than running it on linux?