amymcgovern/pyparrot #173: just want images from VLC video stream

cvenour, 2019-09-02:
I'd like to alter demoMamboVisionGUI.py so that no GUI is drawn to the screen. I just want my python program to have access to the images from the video stream so that I can do some processing on them but I  don't want to display those streaming images to screen. Can someone tell me how to do that? 

Do I comment out certain chunks of code from DroneVisionGUI.py or can I just alter demoMamboVisionGUI.py somehow?

amymcgovern, 2019-09-02:
You can't get rid of the VLC GUI part as it is how it gets the video stream.  The ffmpeg version is your only option if you want no GUI.

cvenour, 2019-09-02:
o.k. thanks for the feedback. And thank you for creating the pyparrot api Amy - it is awesome!!