amymcgovern/pyparrot #174: Connecting to a parrot and a mambo

Yeb02, 2019-09-02:
No issue here, just a question. Is it possible to connect to both a mambo and a bebop with the same laptop (via wifi), and run a single script that commands both of them ?
(With or without an external wifi adapter)

amymcgovern, 2019-09-02:
Not unless you hack one of them to have a fixed IP address.  In fact, you would need to hack both... Because each acts as its own router.

Yeb02, 2019-09-04:
Can't it be done quickly on a bebop ? You certainly know about the hacking guide for bebop and bebop 2, and they make it look like easy. Am I missing something ? 
And if I was to successfully conect a bebop 1 and a bebop 2 to the same computer, is there a way I can discriminate them ? ( i.e. would connecting to the first dron via wifi, then bebop1 = Bebop(), then connect to the second, then bebop2 = Bebop()   work.

amymcgovern, 2019-09-04:
I honestly haven't done it.  I didn't want to break my bebop since I need to support the regular firmware.  I know people who have done it and think it is easy but they have a lot of systems experience.