amymcgovern/pyparrot #175: "Asking for services and characteristics" hanging then disconnecting

ELL4J4NE, 2019-09-09:
I am using a mambo drone and connecting using pyparrot. It successfully connects (full green headlights) then hangs on "asking for services and characteristics" and finally disconnects. I have no sleeps in the code and the same happens when I'm using the (BLE) sample code. Any idea what might be going wrong here?

amymcgovern, 2019-09-09:
you need to post your code and your OS so we can help more

ELL4J4NE, 2019-09-10:
Hi, I'm running Kali linux for raspberry pi and used the mambo claw demo code, thanks

amymcgovern, 2019-09-10:
Ok, so you are using the BLE protocol only and not wifi?

ELL4J4NE, 2019-09-10:
Yes, just BLE

amymcgovern, 2019-09-10:
Could you post the exact printouts from that script?  I haven't run the BLE stuff in a long time but what I see says that the "asking for services" should also print success.

ELL4J4NE, 2019-09-10:
Yes, it prints that it has successfully connected but then fails to carry out any commands and eventually disconnects

amymcgovern, 2019-09-10:
Does it ever print any of the other printouts:

print("connected: %s" % success)

# get the state information
print("sleeping")
mambo.smart_sleep(2)
mambo.ask_for_state_update()
mambo.smart_sleep(2)

print("taking off!")
mambo.safe_takeoff(5)

print("open and close the claw")
mambo.open_claw()
# you have to sleep to let the claw open (it needs time to do it)
mambo.smart_sleep(5)

mambo.close_claw()
# you have to sleep to let the claw close (it needs time to do it)
mambo.smart_sleep(5)

print("landing")
mambo.safe_land(5)
mambo.smart_sleep(5)

print("disconnect")
mambo.disconnect()

amymcgovern, 2019-09-11:
Does it ever print this:
       "magic handshake to make the drone listen to our commandsandsensors"