amymcgovern/pyparrot #177: parrot mambo error (canot import "service browser from zeroconf)

rhj0970, 2019-09-28:
Hi! I was trying to install pyparrot on Rasberry Pi 3 to operate mambo and I finished through the installation guide and when I try to run sample code it gives me an error saying it can't find import service browser from zeroconf. I set up to mambo bluetooth address and set WIFI to false. Also I changed "From pyparrot.mambo to pyparrot.Minidrone since it doesn't have mambo class files.

amymcgovern, 2019-10-05:
Sorry this got lost in my inbox.  we definitely are not using Pyparrot.mambo anymore.  Only Minidrone.  Where is that sample code and I'll fix it.  Also, you shouldn't need zeroconf since that is wifi only.

rhj0970, 2019-10-23:
I was running this code through the python installed in rasberry pi instead of command line.
I got it working now

closed 2019-10-23 by rhj0970