amymcgovern/pyparrot #182: Bebop 2 not connecting, Firewall turned off.

kgpinco, 2019-10-14:
When I connect to my bebop 2 and run this code:

`
"""
Demos the tricks on the bebop. Make sure you have enough room to perform them!
Author: Amy McGovern
"""
from pyparrot.Bebop import Bebop
bebop = Bebop()
print("connecting")
success = bebop.connect(10)
print(success)
print("sleeping")
bebop.smart_sleep(5)
bebop.ask_for_state_update()
bebop.safe_takeoff(10)
print("flip left")
print("flying state is %s" % bebop.sensors.flying_state)
success = bebop.flip(direction="left")
print("mambo flip result %s" % success)
bebop.smart_sleep(5)
print("flip right")
print("flying state is %s" % bebop.sensors.flying_state)
success = bebop.flip(direction="right")
print("mambo flip result %s" % success)
bebop.smart_sleep(5)
print("flip front")
print("flying state is %s" % bebop.sensors.flying_state)
success = bebop.flip(direction="front")
print("mambo flip result %s" % success)
bebop.smart_sleep(5)
print("flip back")
print("flying state is %s" % bebop.sensors.flying_state)
success = bebop.flip(direction="back")
print("mambo flip result %s" % success)
bebop.smart_sleep(5)
bebop.smart_sleep(5)
bebop.safe_land(10)
print("DONE - disconnecting")
bebop.disconnect()
`
It gives this error: 

> connecting
> Setting up mDNS listener since this is not a Mambo
> Making a browser for _arsdk-090c._udp.local.
> ?[38;5;196m connection failed: did you remember to connect your machine to the Drone's wifi network? ?[0m

I have followed the advise on this thread https://github.com/amymcgovern/pyparrot/issues/17 however I have had no luck.

My firewall is not turned on.

I am running Windows 10 and Visual Studio Code. I know the bebop and code run as it works perfectly on my friends laptop.

junupe, 2020-06-21:
I've similar issue. I've tried to connect to my Bebop2 with my Raspberry Pi4 and Macbook pro without success. On mac I've 3.6.x python interpreter and on RPI 3.7.x interpreter. On raspberry the RPI connects on Parrot's wi-fi but Mac does not connect at all on Parrot wi-fi :( Anyway, both ends in the same end-result when I run the test code it cannot connect since it can't find drone's IP for some reason. Any idea why? None of the previous posts were unfortunately working and really would like to get this working since was planning to use this for one project :)

amymcgovern, 2020-06-21:
Sorry I no longer have a functional Bebop 2 to fly and test :(

amymcgovern, 2020-06-21:
Sounds like they upgraded the firmware somehow

junupe, 2020-06-21:
Thanks for your reply Amy. Sad to hear that you don't anymore have Bebop2.
My Bebop2's fw seems to be 4.7.1 is that newer than the latest you've
successfully tested?
BR,
junupe

amymcgovern, 2020-06-21:
I have two bebop2's but my battery charger died and so no functional batteries :( So I can't even turn it on to check!