amymcgovern/pyparrot #183: Bluetooth LE Under Windows

AlexanderDavid, 2019-10-24:
I'm having trouble interfacing with the Mambo drone (sans FPV so I can't use WiFi) over BLE. 

Have there been any attempts to get Bluetooth LE working under windows for the Mambo? I noticed that it has been referenced in a couple of other issues that the main factor holding back development was a lack of a stable platform. I think that the [Bleak](https://github.com/hbldh/bleak) library might work. I am going to try and get some stuff working but I just wanted to ask to see if this has been attempted before.

HassanFarag, 2020-07-14:
@AlexanderDavid Did you succeed to port the BLE to windows using Bleak? Please let me know! Thanks in advance

AlexanderDavid, 2020-07-14:
@HassanFarag unfortunately not. I don't know the current state of windows Bluetooth LE support but at the time it was severely lacking. The route I decided to go with was to buy a Bluetooth dongle that one library had recommended but I graduated before I was able to work on it. I would recommend just using Linux if that's possible in your situation.