amymcgovern/pyparrot #184: Issue pertaining bluepy permission during findMambo.py

InquisitiveAS, 2019-11-04:
This is what I get, can anyone know what is the issue
``$ sudo python findMinidrone.py
Traceback (most recent call last):
  File "findMinidrone.py", line 49, in <module>
    main()
  File "findMinidrone.py", line 30, in main
    devices = scanner.scan(10.0)
  File "/usr/local/lib/python2.7/dist-packages/bluepy/btle.py", line 852, in scan
    self.start(passive=passive)
  File "/usr/local/lib/python2.7/dist-packages/bluepy/btle.py", line 790, in start
    self._mgmtCmd("le on")
  File "/usr/local/lib/python2.7/dist-packages/bluepy/btle.py", line 309, in _mgmtCmd
    rsp = self._waitResp('mgmt')
  File "/usr/local/lib/python2.7/dist-packages/bluepy/btle.py", line 366, in _waitResp
    raise BTLEManagementError("Management not available (permissions problem?)", resp)
bluepy.btle.BTLEManagementError: Management not available (permissions problem?)

InquisitiveAS, 2019-11-04:
I also cross checked with the bluepy installation but found issues!

$ sudo pip install bluepy
The directory '/home/.cache/pip/http' or its parent directory is not owned by the current user and the cache has been disabled. Please check the permissions and owner of that directory. If executing pip with sudo, you may want sudo's -H flag.
The directory '/home/.cache/pip' or its parent directory is not owned by the current user and caching wheels has been disabled. check the permissions and owner of that directory. If executing pip with sudo, you may want sudo's -H flag.
Requirement already satisfied: bluepy in /usr/local/lib/python2.7/dist-packages

InquisitiveAS, 2019-11-04:
I installed python 3 but got the same bluepy issue

amymcgovern, 2019-11-04:
I don't maintain the bluepy package.  You will want to post there.

InquisitiveAS, 2019-11-04:
Thanks a lot