amymcgovern/pyparrot #186: Working with Parrot Mambo Firmware 3.0.17

guigabyte-koko, 2019-12-04:
I am trying to process ground cam pictures from the Parrot Mambo Fly. I see that the firmware is broken for 3.0.25 and 3.0.26. I found older versions of the firmware here:
[](https://www.drone-forum.com/forum/viewtopic.php?f=127&t=10120)

I have two questions:

1.  Once the older firmware is on the Mambo, what do I need to edit in order to be able to process ground cam images through BLE connection?

2. I am having trouble getting the older firmware onto the Mambo, is there anything special I need to do to get an older version on?

I am using these instructions at the bottom of this page:

[](https://www.parrot.com/us/support/products/parrot-mambo/software)

Any help would be appreciated!

amymcgovern, 2019-12-04:
I have not installed the older firmware.  For a long time, they had the firmware removed from the parrot pages so it wasn't possible.  Once the firmware is installed, the ground cam images should simply work however.

guigabyte-koko, 2019-12-04:
I am able to connect and fly the Mambo with firmware 3.0.25. However, I have another Parrot Mambo with Firmware version 3.0.18 and I am able to connect to it but it will not even take off. If I run DemoMamboDirectFlight it will print that the connection succeeded and will print that it is taking off and landing, but the entire time the drone is stationary. I am running the same script as I was running using firmware 3.0.25, and only changing the address for the BLE connection. 

Should I try to branch off of an older version of pyparrot? Or should the newest commit still be compatible with older firmware?

amymcgovern, 2019-12-05:
It *should* be but I honestly haven't run it recently.  I've switched drone platforms due to parrot dropping their support.  But I didn't upgrade some of my older drones entirely to ensure compatibility and they were all running last time I flew them in an event (probably 4 months ago, before I swapped platforms).

kontrafiktion, 2019-12-21:
Which platform are you using now? Is there already some code?

amymcgovern, 2019-12-27:
I'm using Tello edu's.