amymcgovern/pyparrot #193: IA Class

malfonsoNeoris, 2020-01-29:
hi amy, love your library
I am working with drones and i'm getting curious about the IA class you give with this drones. i have read the assignements, and would love to DO them...  specially IA Agents, and reinforced learning.
Is there any course material, book names from where i can get the knowledge neccessary in order to do the asignements ( IA Agents for example). or if you provide or recommend a online course that i can take

thanks..

PD: Sorry if this is not the place to make this question

amymcgovern, 2020-01-29:
I am using the AIMA textbook by Russell & Norvig.  It is an excellent resource!  There are online classes in AI though most focus on learning.  I teach the broader AI in this class since I also teach a focused ML class.

malfonsoNeoris, 2020-01-30:
thanks so much..

closed 2020-01-30 by malfonsoNeoris