amymcgovern/pyparrot #195: parrot mambo discontinued

cvenour, 2020-02-14:
Hi Amy,

As you probably know, the Parrot Mambo is being discontinued: http://droneadvisor.tech/news/parrot-plans-to-stop-selling-mambo-and-swing-drones

This is too bad because it's an excellent and affordable drone and the pyparrot API is great.

Do you know of a good alternative to this drone that has an API/SDK? For example what kind of drone will you be using in future classes?

amymcgovern, 2020-02-14:
I have switched to the Tello Edu.  I also have a library for it although it is in alpha mode right now but we are using it in my class this semester!

cvenour, 2020-02-14:
That's great news - thank you. Once my parrot drone has _________ <monty python euphemism for dead parrot> I'll get the Tello Edu and start using your api. Thanks very much for creating this library!