amymcgovern/pyparrot #197: Example code

lucasantoro, 2020-03-12:
Good evening,
Where are the code for ai_class and workshop? 
I'm very interested in study these applications as student attending a course of robotics engineering.
Thank you very much for consideration and kindness

amymcgovern, 2020-03-14:
https://github.com/amymcgovern/pyparrot/tree/master/coursework

lucasantoro, 2020-03-14:
Good morning,
I read all documents, but the code related to these task where are located?
In any case, i'm using 4 mambo, do you have some suggestion to realize a swarm of these little drone?
Thank you very much for your consideration and kindness

amymcgovern, 2020-03-17:
they just relied on the library itself and wrote their own code, which is not checked in.

Check out https://github.com/amymcgovern/pytello/ also.  It will handle swarms easier also since Tello lets you choose IP addresses and connect all to one router.  I'm developing that library though there are other Tello python libraries out there also.

lucasantoro, 2020-03-22:
Thank you very much for your consideration.
I'm checking out your new library.
Do you think that is possible to build a waypoint follower drone with mambo parrot and python library?
Anyway, to satisfy my curiosity, how can you develope this library? Are you stareted from parrot's SDK from github? I want to learn the proper way to approach at this "new  world".
Thank you very much again for you kindness

amymcgovern, 2020-03-23:
Yes, you can use the vision system to follow waypoints!