amymcgovern/pyparrot #199: timeout error - try again [Bebop 2]

tommy1223, 2020-05-04:
I am doing a project using Bebop 2. I am using Raspberry Pi to control the drone. After I take off the drone, I have a 3 minutes algorithm for the drone so it has to hover 3 minutes without moving. However, it sometimes returns timeout error in a random time and I lose all the control to the drone. I am using takeoff() and land() function directly so I think it should be not related to the takeoff or land timeout. The error is not always happening and it happens randomly at different times of my program.

Is there a solution regarding this? Or is it possible to disable the timeout feature?

Thank you in advance!

amymcgovern, 2020-05-05:
The timeout is from the connection itself.  Can you do a try/catch with the connection and reconnect?

R0M107, 2024-06-21:
Hi,
I don't know if this project is still maintained or not but I have the same problem, it say :
timeout - trying again
At the time I start the program, like 2 second after, this message is displayed and after it is displayed every 5 seconds.
I don't know how to solve this problem.

Sorry if my english is bad, I'm french.

amymcgovern, 2024-06-22:
I'm still here but I don't have any running parrot drones anymore :(