amymcgovern/pyparrot #202: ip address can not obtained in linux usb lancard with bebop2

freaad, 2020-07-08:
Hello.
I am recently try to run the bebop code again.
So the code itself and bebop is working well in the window pc.
I have workstation in the lab so I want to apply your code to my drone.
I face this error.
Do you have any idea about it?

Thank you.

(torch) kangd345@ce-yc-dlearn2:~$ env PTVSD_LAUNCHER_PORT=37199 /home/kangd345/anaconda2/envs/torch/bin/python /home/kangd345/.vscode/extensions/ms-python.python-2020.2.64397/pythonFiles/lib/python/new_ptvsd/no_wheels/ptvsd/launcher /home/kangd345/drone_code_test/pyparrot-master/demoBebopIndoors.py 
connecting
Setting up mDNS listener since this is not a Mambo
Making a browser for _arsdk-090c._udp.local.
Service Bebop2Power-023404._arsdk-090c._udp.local. added, service info: ServiceInfo(type='_arsdk-090c._udp.local.', name='Bebop2Power-023404._arsdk-090c._udp.local.', addresses=[b'\xc0\xa8*\x01'], port=44444, weight=0, priority=0, server='Bebop2Power-023404.local.', properties={b'{"device_id":"PI040436AA7K023404"}': None})
Traceback (most recent call last):
  File "/home/kangd345/drone_code_test/pyparrot-master/demoBebopIndoors.py", line 14, in <module>
    success = bebop.connect(10)
  File "/home/kangd345/drone_code_test/pyparrot-master/pyparrot/Bebop.py", line 224, in connect
    connected = self.drone_connection.connect(num_retries)
  File "/home/kangd345/drone_code_test/pyparrot-master/pyparrot/networking/wifiConnection.py", line 171, in connect
    handshake = self._handshake(num_retries)
  File "/home/kangd345/drone_code_test/pyparrot-master/pyparrot/networking/wifiConnection.py", line 318, in _handshake
    self.drone_ip = ipaddress.IPv4Address(self.connection_info.address).exploded
AttributeError: 'ServiceInfo' object has no attribute 'address'

amymcgovern, 2020-07-08:
Race condition maybe in what the bebop sent back?  Does it happen every time?  Looks like a bad packet to me.

freaad, 2020-07-16:
> Race condition maybe in what the bebop sent back? Does it happen every time? Looks like a bad packet to me.
I realize that it only happen in school computer. It might be the school firewall problem.

closed 2020-07-18 by freaad