amymcgovern/pyparrot #203: AttributeError: 'project' has no attribute 'myclass'

mipcmhone6s, 2020-08-02:
[image]

Hello!
I ran "demoMamboDirectFlight.py", but I met this issue.

Does anyone have solved it?

OhSeungBeen, 2020-08-03:
I solved it myself.
I changed class -> myclass in xml.

Thanks.

toncho11, 2022-11-02:
> Hi, it seems like this is related to untangle updates. You will need to edit: at least line 163 in DroneSensorParser.py and 45 in DroneCommandParser, if you take a look you will note that where my_file.project.myclass need to be replace by my_file.project.get_elements() in order to align to new untangle parser object. Best.

Yes. This fixed the problem, but why is it not updated in the main code? Maybe because there is a suggestion for another XML parsing library by ShusenLiu?

amymcgovern, 2022-11-02:
I haven't updated as I haven't been supporting the project anymore due to not having working parrot drones anymore (since parrot stopped supporting them)

toncho11, 2022-11-03:
I see. You mean the Minidrones are no longer supported?

amymcgovern, 2022-11-03:
Yes, parrot stopped supporting anything except the anafi